raspberrypi/Raspberry-Pi-OS-64bit #159: HDD problem

Anenemy1, 2021-06-23:
Hello friends and supporters,
I had a RPI4B 8GB with argon one v2 case and pi os arm64 latest build of 07/05/2021 on a sata ssd 256GB. I also have an official argon power supply of 4.25V and 3,5A. My problem is everytine i try to connect my external hdd, it frozes, its SSD disconnected, and while usb mouse and keyboard powered have no input. I also tryed with ssh but it can't connect. The only solution is to hard poweoff. If i restart i need to disconnect the HDD because i 'll have no boot.

lurch, 2021-06-23:
What does this have to do with Raspberry Pi OS 64bit? Sounds more like a question for the https://www.raspberrypi.org/forums ?
(sounds like a lack-of-power problem)

Anenemy1, 2021-06-23:
Ok. I 'll delete this issue and ask them.

closed 2021-06-23 by Anenemy1