raspberrypi/Raspberry-Pi-OS-64bit #161: "Input not supported" message on console after regular boot

UgoPoddine, 2021-06-26:
Hello everyone,

on a RaspberryPI3 running well with PI-OS 32 bits I started the OS64bit image downloaded from here.
The Raspberry starts as expected, I can see on the console screen the first messages (e.g. the SD card resize message) and the autologin message. After this the screen goes in "Input not supported" error.
Take care that :

a) the Raspberry is running correctly and I can access it via SSH (I'm just doing a system update)
b) the same issue happens to me both using Debian ARM64 image and Ubuntu ARM64 image

My physical console screen is a not that new : is ACER AL1716 fed by an HDMI 2 analog converter, perfectly running when the image is 32 bit.

Can you help in solving ?
Thank-you very much
Best Regards
Ugo Poddine

CPU INFO

Hardware        : BCM2835
Revision        : a02082
Serial          : 
Model           : Raspberry Pi 3 Model B Rev 1.2

UgoPoddine, 2021-06-28:
Hello,

just for adding a remark.
I have also tried to force screen resolution at boot, because I have seen in fbset -s that the default set during ARM64 boot is not the same of ARM32 and my screen resolution in running scenario io 1280x1024.

framebuffer_width=1280
framebuffer_height=1024

in config.txt
Unfortunately this seems not to solve the issue : 

fbset -s   is now

mode "1280x1024"
    geometry 1280 1024 1280 1024 16
    timings 0 0 0 0 0 0 0
    accel true
    rgba 5/11,6/5,5/0,0/0
endmode

.but still "Input not supported" is there

Regards
UP

pelwell, 2021-06-28:
Do you have a means of editing `config.txt` in the event that you end up breaking your boot (a PC with a card reader, a separate Pi image and a card reader, etc.)? If so, please edit config.txt in your _32-bit_ card and add `dtoverlay=vc4-kms-v3d` and try booting it.

If that image also fails to show anything on your display then we know that this is not a problem specific to the 64-bit image (which uses the `vc4-kms-v3d` overlay by default).

6by9, 2021-06-28:
Please report which kernel version you are running. ```uname -a```

UgoPoddine, 2021-06-28:
Hello 6by9,

I beg your pardon : 

Linux raspberrypi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aar                                                                                                       ch64 GNU/Linux

Best regards
Ugo

UgoPoddine, 2021-06-28:
Hello Pelwell,

if I understand correctly I have to : 
a) create a copy of the 32-bit image, in case of corruption during the test
b) change the config.txt of the 32 bit image adding dtoverlay=vc4-kms-v3d
c) trying to reboot 

If he effect on the 32 bit image is the same of that on the 64 bit, the issue is in c4-kms-v3d and not realted to the 64bit image.
Please confirm

Thank-you
Ugo

pelwell, 2021-06-28:
Yes, that's about right. However, I think you can skip step a) if you have a way of removing the dtoverlay line when the display doesn't work - another computer with an SD card reader, using `ssh` or a serial connection to connect to the Pi, etc. - I really don't think running this test will corrupt your card.

UgoPoddine, 2021-06-28:
Hello Pelwell,

The test is quite interesting.
My current 32bit image config.txt says : 

[...]

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[...]

I have modified it in : 

[...]

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

dtoverlay=vc4-kms-v3d

[...]

.. and rebooted.
The Rasperry has started correctly : in console mode (on the VGA adapter) i have reached the command line (I'm starting in text mode) without any issue, differently from what happened on 64 bit image.
But when I performed the startx command, the "input not supported message" is again there also on 32bits.
Than your hypothesis seems to be right and the issue is hidden in the "not fale" kms....
I wonder that the preinstalled 64 bits image is trying to start directly in X mode ? Perhaps for being sure I can starts in command line mode also the 64bits image, for comparing the effect...

Ugo

UgoPoddine, 2021-06-28:
Hi all,

I have started the ARM64 image using the defult config.txt file (that is with  : 

[...]

# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1

[...]

... but without starting the X server at boot.
Doing this, the situation on ARM64 is the same of 32bit image : I can get the command line on the console screen without any "input not supported" error.
If I execute startx command, "input not supported" is there.
If I understand correctly : 

a) on 32bit : i can not use dtoverlay=vc4-kms-v3d
b) on 64bit : I can not use dtoverlay=vc4-fkms-v3d

.. when running X server.
Must I try to start 64bit image with : 

dtoverlay=vc4-kms-v3d

?

Ugo

lurch, 2021-06-28:
> My current 32bit image `config.txt` says :
```
[pi4]

dtoverlay=vc4-fkms-v3d
max_framebuffers=2
```
> I have modified it in :
```
[pi4]

#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

dtoverlay=vc4-kms-v3d
```

The `[pi4]` line indicates a [conditional filter](https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md), and as you're using a "Model : Raspberry Pi 3 Model B Rev 1.2" anything inside a `[pi4]` section won't have any affect. So you should add a `[all]` line to "turn off" the `pi4` filter, and then add your changes after that.

UgoPoddine, 2021-07-10:
Hi all,

begging your pardon for the delay in answering, due to a job travel, I have performed the suggested test on the 32bit image, by modifying the config.txt as follow, starting 32bit image on the same machine / hardware with  dtoverlay=vc4-kms-v3d
The effect is the one you expected : the 32bit machine start till command line on console, but at startx I get the same ARM64 "input not supported" errror.
Tha machine is in any case still active in headless mode (SSH or VNC)
Must I try to use vc4-fkms-v3d on 64bit scenario ?

Please advise
Thank-you
Regards
Ugo

[....]

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=vc4-kms-v3d

[....]

UgoPoddine, 2021-07-13:
Hello,

I was finally able to solve the problem.
For doing this I have modified the config.txt on the ARM64 image preventing the use both of vc4-fkms-v3d and of vc4-kms-v3d, that is starting the machine in the same way of what the 32bit image is doing.

[...]

[all]
#dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-kms-v3d

[...]

I'm not sure if this can generate issues, but at the moment the startx command runs as expected and the X environment is accessible also from the console

Thank-you
Best regards
Ugo

closed 2021-07-13 by UgoPoddine