raspberrypi/Raspberry-Pi-OS-64bit #162: ffmpeg can't decode with hw HEVC decoder

xlazom00, 2021-07-04:
When I run this simple command
ffmpeg -hwaccel drm -v debug -i TearsOfSteel_720p_h265.mkv -f null -

It looks like ffmpeg really use hevc hw decoder
[hevc @ 0x5578c381f0] Trying V4L2 devices: /dev/media2,/dev/video19
but
[hevc @ 0x5578c5e2e0] Output frame with POC 3.
[hevc @ 0x5578c5eeb0] nal_unit_type: 0(TRAIL_N), nuh_layer_id: 0, temporal_id: 0
[AVHWFramesContext @ 0x5578c3fcc0] drm_transfer_data_from: Copy fail
[hevc @ 0x5578c191e0] Failed to transfer data to output frame: -22.
Error while processing the decoded data for stream #0:0

All works fine with 32bit version of OS

suqin-haha, 2022-03-17:
I also got trouble for playing x265 video, after install 64bit version OS
works fine in 32bit OS before.  both vlc and kodi cannot work.

lurch, 2022-03-18:
@popcornmix ?

popcornmix, 2022-03-19:
Kodi works fine with hevc hw decode on 64-bit for me.
Will need debug enabled kodi.log file to diagnose further.

suqin-haha, 2022-03-19:
Here is the Kodi log file, Thank you for the checking.
[kodi.log](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8310045/kodi.log)

In the Kodi log, I try to run 
1. [jellyfish-3-mbps-hd-hevc.mkv](https://jell.yfish.us/media/jellyfish-3-mbps-hd-hevc.mkv),  can play but choppy
2. [jellyfish-3-mbps-hd-hevc-10bit.mkv](https://jell.yfish.us/media/jellyfish-3-mbps-hd-hevc-10bit.mkv), cannot even open it
I used the test file from: https://jell.yfish.us/

What I can notice is that:
when small file, it may playable normal HEVC file but choppy, and cannot play10 bit HEVC file 
when it comes larger, it's more worse.

for vlc, it even shows crushed image and choppy

suqin-haha, 2022-03-19:
> Kodi works fine with hevc hw decode on 64-bit for me. Will need debug enabled kodi.log file to diagnose further.

Here is the Kodi log file, Thank you for the checking.
[kodi.log](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8310045/kodi.log)

In the Kodi log, I try to run 
1. [jellyfish-3-mbps-hd-hevc.mkv](https://jell.yfish.us/media/jellyfish-3-mbps-hd-hevc.mkv),  can play but choppy
2. [jellyfish-3-mbps-hd-hevc-10bit.mkv](https://jell.yfish.us/media/jellyfish-3-mbps-hd-hevc-10bit.mkv), cannot even open it
I used the test file from: https://jell.yfish.us/

What I can notice is that:
when small file, it may playable normal HEVC file but choppy, and cannot play10 bit HEVC file 
when it comes larger, it's more worse.

for vlc, it even shows crushed image and choppy

popcornmix, 2022-03-20:
Make sure in config.txt you have hevc decoder overlay:
```dtoverlay=rpivid-v4l2```.
Also for 4K you likely need to increase the cma allocation
```dtoverlay=vc4-kms-v3d,cma-512```

suqin-haha, 2022-03-20:
> Make sure in config.txt you have hevc decoder overlay: `dtoverlay=rpivid-v4l2`. Also for 4K you likely need to increase the cma allocation `dtoverlay=vc4-kms-v3d,cma-512`

Great!! now it's works fine after add the `dtoverlay=rpivid-v4l2`
um... just wonder, if just add one line will solve the problem, why it not just come with the 64bit system? or maybe my system is an odd case?
Anyway, Thank you so much for the help!

popcornmix, 2022-03-21:
The original hevc decoder didn't use v4l2 but a simpler (hackier) interface to the hardware.
The v4l2 driver was added as an option (through a dtoverlay), as it was developed and debugged.

We are at the point where the older driver is deprecated and the v4l2 is the recommended, so we are looking at changing the default. However there may be use cases that expect the older driver (especially if they are not using the kms display driver).

suqin-haha, 2022-03-21:
> The original hevc decoder didn't use v4l2 but a simpler (hackier) interface to the hardware. The v4l2 driver was added as an option (through a dtoverlay), as it was developed and debugged.
>
> We are [...]

yeah, I see that there are some trade-offs, Thanks for the patient answer

suqin-haha, 2022-04-02:
Hi @popcornmix ,

Here is new problem that I cannot use Bluetooth speaker with Kodi, the Bluetooth speaker is good for others.

My observation:
the Bluetooth speaker will loss connection when I open Kodi... then there is no sound.
However, my Bluetooth mouse doesn't loss connection (Or it reconnects too fast to notice).
And after I quit Kodi, if I click the tray area, like the volume button or the Bluetooth button, the tray area will refresh and reconnect the Bluetooth device.

Log file:
[kodi.log](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8403485/kodi.log)

Environment is same, still raspberry pi 4 with 64bit Pi OS, and as I remember it works no problem on 32bit.
one of test file I use is: [jellyfish-55-mbps-hd-h264.mkv](https://jell.yfish.us/media/jellyfish-55-mbps-hd-h264.mkv)

Thanks!

popcornmix, 2022-04-03:
@suqin-haha please create a new issue and don't hijack an unrelated one.