raspberrypi/Raspberry-Pi-OS-64bit #163: Lately, getting this error on my RPi 64bit. Appreciate help fixing it.

bislink, 2021-07-08:
```
pi@vpn2:~ $ sudo apt update && sudo apt upgrade
Hit:1 http://deb.debian.org/debian buster InRelease
Get:2 http://deb.debian.org/debian-security buster/updates InRelease [65.4 kB] 
Get:3 https://download.docker.com/linux/debian buster InRelease [54.0 kB]      
Get:4 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]          
Get:5 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]         
Hit:6 https://packages.gitlab.com/gitlab/raspberry-pi2/debian buster InRelease 
Get:7 http://archive.raspberrypi.org/debian buster/main armhf Packages [376 kB]
Get:8 http://archive.raspberrypi.org/debian buster/main arm64 Packages [287 kB]
Fetched 866 kB in 3s (282 kB/s)   
Reading package lists... Done
Building dependency tree       
Reading state information... Done
4 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  linux-image-4.19.0-17-arm64 raspberrypi-sys-mods raspberrypi-ui-mods
  rpi-eeprom
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
Need to get 1,036 kB/42.0 MB of archives.
After this operation, 528 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.raspberrypi.org/debian buster/main arm64 rpi-eeprom arm64 12.9-1 [701 kB]
Get:2 http://archive.raspberrypi.org/debian buster/main arm64 raspberrypi-ui-mods arm64 1.20210706 [323 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main arm64 raspberrypi-sys-mods arm64 20210706 [12.2 kB]
Fetched 1,036 kB in 2s (584 kB/s)               
Reading changelogs... Done
(Reading database ... 240148 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_12.9-1_arm64.deb ...
Unpacking rpi-eeprom (12.9-1) over (12.8-1) ...
Preparing to unpack .../linux-image-4.19.0-17-arm64_4.19.194-2_arm64.deb ...
Unpacking linux-image-4.19.0-17-arm64 (4.19.194-2) over (4.19.194-1) ...
dpkg: error processing archive /var/cache/apt/archives/linux-image-4.19.0-17-arm64_4.19.194-2_arm64.deb (--unpack):
 unable to make backup link of './boot/System.map-4.19.0-17-arm64' before installing new version: Operation not permitted
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../raspberrypi-ui-mods_1.20210706_arm64.deb ...
Unpacking raspberrypi-ui-mods (1.20210706) over (1.20201210+nmu1) ...
Preparing to unpack .../raspberrypi-sys-mods_20210706_arm64.deb ...
Unpacking raspberrypi-sys-mods (20210706) over (20210310) ...
Errors were encountered while processing:
 /var/cache/apt/archives/linux-image-4.19.0-17-arm64_4.19.194-2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@vpn2:~ $ 
```

lurch, 2021-07-08:
> linux-image-4.19.0-17-arm64

That's kinda odd, I dunno why `apt` would be trying to install such an old kernel version. Have you done some strange customisation of your `sources.list`? Are you able to reproduce this problem with a fresh image?

bislink, 2021-07-08:
I did not make any changes to sources.list manually.

Will try to see if i can reproduce the same error with an(other) SD card

bislink, 2021-07-08:
Here is my sources.list

pi@vpn2:~ $ cat /etc/apt/sources.list
deb http://deb.debian.org/debian buster main contrib non-free
deb http://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb http://deb.debian.org/debian buster-updates main contrib non-free
# Uncomment deb-src lines below then 'apt-get update' to enable 'apt-get source'
#deb-src http://deb.debian.org/debian buster main contrib non-free
#deb-src http://deb.debian.org/debian-security/ buster/updates main contrib non-free
#deb-src http://deb.debian.org/debian buster-updates main contrib non-free
pi@vpn2:~ $

lurch, 2021-07-08:
I think there might also be other files in the `/etc/apt/sources.list.d` folder? I spotted `https://download.docker.com/linux/debian` and `https://packages.gitlab.com/gitlab/raspberry-pi2/debian` in your log above...

bislink, 2021-07-08:
pi@vpn2:~ $ sudo ls /etc/apt/sources.list.d/
docker.list  gitlab_raspberry-pi2.list	raspi.list  vscode.list
pi@vpn2:~ $

XECDesign, 2021-07-08:
Our kernel comes from the `raspberrypi-kernel` package. What you're installing is Debian's kernel package, which won't work.

bislink, 2021-07-08:
pi@vpn2:~ $ sudo cat /etc/apt/sources.list.d/raspi.list 
deb http://archive.raspberrypi.org/debian/ buster main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ buster main
pi@vpn2:~ $

bislink, 2021-07-08:
So, what should I do. I did not make these changes manually. 

Should I disable all except raspi.list?

XECDesign, 2021-07-08:
This isn't related to your package lists. First, take a backup. Then run the following:

```
sudo apt purge linux-image-4.19.0-17-arm64 -y
sudo apt install raspberrypi-kernel -y
```

If it says `raspberrypi-kernel` is already installed, add `--reinstall` to that second command.

bislink, 2021-07-08:
Output from first command - No errors

pi@vpn2:~ $ sudo apt purge linux-image-4.19.0-17-arm64 -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  linux-image-4.19.0-17-arm64* linux-image-arm64*
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
5 not fully installed or removed.
After this operation, 227 MB disk space will be freed.
(Reading database ... 240145 files and directories currently installed.)
Removing linux-image-arm64 (4.19+105+deb10u12) ...
Removing linux-image-4.19.0-17-arm64 (4.19.194-1) ...
/etc/kernel/postrm.d/initramfs-tools:
update-initramfs: Deleting /boot/initrd.img-4.19.0-17-arm64
Setting up raspberrypi-sys-mods (20210706) ...
Setting up dillo (3.0.5-5+rpt1) ...
Setting up raspberrypi-ui-mods (1.20210706) ...
Setting up rpi-update (20210618) ...
Setting up rpi-eeprom (12.9-1) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for shared-mime-info (1.10-1) ...

Processing triggers for desktop-file-utils (0.23-4) ...
(Reading database ... 236290 files and directories currently installed.)
Purging configuration files for linux-image-4.19.0-17-arm64 (4.19.194-1) ...
pi@vpn2:~ $ sudo apt install raspberrypi-kernel -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20210527-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

bislink, 2021-07-08:
Working on the second one with --reinstall

bislink, 2021-07-08:
60% done

bislink, 2021-07-08:
Done with some warnings.

```
pi@vpn2:~ $ sudo apt install raspberrypi-kernel -y --reinstall
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/21.5 MB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 236293 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel_1%3a1.20210527-1_arm64.deb ...
run-parts: executing /etc/kernel/prerm.d/dkms 5.10.17+ /boot/kernel.img
run-parts: executing /etc/kernel/prerm.d/dkms 5.10.17-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/prerm.d/dkms 5.10.17-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/prerm.d/dkms 5.10.17-v8+ /boot/kernel8.img
Leaving 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'
Leaving 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelhack'
Leaving 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernelhack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-plus.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-b-rev1.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-rev1.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-b.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-cm.dtb to /usr/share/rpikernelhack/bcm2708-rpi-cm.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-zero-w.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero-w.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2708-rpi-zero.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero.dtb by rpikernelhack'
Leaving 'diversion of /boot/bcm2709-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2709-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b-plus.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-cm3.dtb to /usr/share/rpikernelhack/bcm2710-rpi-cm3.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-4-b.dtb to /usr/share/rpikernelhack/bcm2711-rpi-4-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-400.dtb to /usr/share/rpikernelhack/bcm2711-rpi-400.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-cm4.dtb to /usr/share/rpikernelhack/bcm2711-rpi-cm4.dtb by rpikernelhack'
Adding 'diversion of /boot/COPYING.linux to /usr/share/rpikernelhack/COPYING.linux by rpikernelhack'
Adding 'diversion of /boot/overlays/README to /usr/share/rpikernelhack/overlays/README by rpikernelhack'
Adding 'diversion of /boot/overlays/act-led.dtbo to /usr/share/rpikernelhack/overlays/act-led.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adafruit18.dtbo to /usr/share/rpikernelhack/overlays/adafruit18.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau1977-adc.dtbo to /usr/share/rpikernelhack/overlays/adau1977-adc.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau7002-simple.dtbo to /usr/share/rpikernelhack/overlays/adau7002-simple.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1015.dtbo to /usr/share/rpikernelhack/overlays/ads1015.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1115.dtbo to /usr/share/rpikernelhack/overlays/ads1115.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads7846.dtbo to /usr/share/rpikernelhack/overlays/ads7846.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv7282m.dtbo to /usr/share/rpikernelhack/overlays/adv7282m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv728x-m.dtbo to /usr/share/rpikernelhack/overlays/adv728x-m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/akkordion-iqdacplus.dtbo to /usr/share/rpikernelhack/overlays/akkordion-iqdacplus.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-boss-dac-pcm512x-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-boss-dac-pcm512x-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-boss2-dac-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-boss2-dac-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-digione.dtbo to /usr/share/rpikernelhack/overlays/allo-digione.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-katana-dac-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-katana-dac-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-piano-dac-pcm512x-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-piano-dac-pcm512x-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-piano-dac-plus-pcm512x-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-piano-dac-plus-pcm512x-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/anyspi.dtbo to /usr/share/rpikernelhack/overlays/anyspi.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/apds9960.dtbo to /usr/share/rpikernelhack/overlays/apds9960.dtbo by rpikernelhack'
[... cut ...]
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 5.10.17-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/apt-auto-removal 5.10.17-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/dkms 5.10.17-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 5.10.17-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/pm-utils 5.10.17-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/unattended-upgrades 5.10.17-v8+ /boot/kernel8.img
pi@vpn2:~ $ 
```

XECDesign, 2021-07-08:
Yup, that looks okay.

bislink, 2021-07-08:
Should I run apt update and upgrade now?

bislink, 2021-07-08:
Or install linux-headers-5.10.17+ first?

XECDesign, 2021-07-08:
No, do not try to install anything like that.

I'm not sure what state your system is in right now, but something like this should get it back to normal.

```
sudo apt install -f
sudo apt update
sudo apt full-upgrade
```

bislink, 2021-07-08:
The above 3 commands : 

```
pi@vpn2:~ $ sudo apt install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@vpn2:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease                                                              
Get:3 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]                                                             
Get:4 https://download.docker.com/linux/debian buster InRelease [54.0 kB]                                                         
Hit:5 http://archive.raspberrypi.org/debian buster InRelease                                                                      
Hit:6 https://packages.gitlab.com/gitlab/raspberry-pi2/debian buster InRelease                                                    
Fetched 106 kB in 2s (49.6 kB/s)                                                              
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
pi@vpn2:~ $ sudo apt full-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@vpn2:~ $ 
```

XECDesign, 2021-07-08:
Then you should be okay now.

bislink, 2021-07-08:
Thank you very much XECDesign and lurch.

AlanNevill, 2022-07-27:
Hi,
I am having the same issue today. apt update reports an old kernel 5.10.127-2 is available.

`alan@raspberrypi:~ $ !478
sudo apt list --upgradable
Listing... Done
linux-image-5.10.0-16-arm64/stable-security 5.10.127-2 arm64 [upgradable from: 5.10.127-1]
N: There is 1 additional version. Please use the '-a' switch to see it
`
Any ideas of how I can prevent this old update from being offered?

Many thanks
Alan

lurch, 2022-07-27:
According to what @XECDesign wrote previously, you shouldn't have any `linux-image` package installed.