raspberrypi/Raspberry-Pi-OS-64bit #165: 'keyboard-configuration' is missing final newline

heX16, 2021-07-09:
I have installed a new clean OS. Version: 2021-05-07-raspios-buster-armhf-lite.img
On `sudo apt upgrade` I get an error.

dpkg: unrecoverable fatal error, aborting:
 files list file for package 'keyboard-configuration' is missing final newline

```
pi@raspberrypi:~ $ sudo apt update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [376 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Fetched 13.5 MB in 13s (1,068 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
33 packages can be upgraded. Run 'apt list --upgradable' to see them.

pi@raspberrypi:~ $ sudo apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
The following packages will be upgraded:
  base-files firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek isc-dhcp-client isc-dhcp-common klibc-utils libgcrypt20 libgnutls30 libhogweed4
  libklibc liblz4-1 libnettle6 libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libx11-6 libx11-data libxml2 linux-libc-dev pi-bluetooth python-gpiozero
  python-spidev python3-gpiozero python3-spidev raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods rpi-eeprom rpi-update
33 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 138 MB of archives.
After this operation, 5,093 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf firmware-atheros all 1:20190114-2+rpt1 [4,012 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf firmware-brcm80211 all 1:20190114-2+rpt1 [4,600 kB]
Get:2 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf base-files armhf 10.3+rpi1+deb10u10 [70.2 kB]
Get:3 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libgcrypt20 armhf 1.8.4-5+deb10u1 [500 kB]
Get:10 http://archive.raspberrypi.org/debian buster/main armhf firmware-libertas all 1:20190114-2+rpt1 [3,424 kB]
Get:13 http://archive.raspberrypi.org/debian buster/main armhf firmware-misc-nonfree all 1:20190114-2+rpt1 [3,341 kB]
Get:16 http://archive.raspberrypi.org/debian buster/main armhf firmware-realtek all 1:20190114-2+rpt1 [506 kB]
Get:17 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-doc armhf 1:1.20210527-1 [31.4 MB]
Get:4 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libnettle6 armhf 3.4.1-1+deb10u1 [223 kB]
Get:5 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libhogweed4 armhf 3.4.1-1+deb10u1 [130 kB]
Get:7 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libgnutls30 armhf 3.6.7-4+deb10u7 [1,049 kB]
Get:8 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf liblz4-1 armhf 1.8.3-1+deb10u1 [49.7 kB]
Get:9 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf isc-dhcp-client armhf 4.4.1-2+deb10u1 [295 kB]
Get:11 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf isc-dhcp-common armhf 4.4.1-2+deb10u1 [144 kB]
Get:12 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf klibc-utils armhf 2.0.6-1+rpi1+deb10u1 [87.1 kB]
Get:14 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libklibc armhf 2.0.6-1+rpi1+deb10u1 [52.8 kB]
Get:15 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libx11-data all 2:1.6.7-1+deb10u2 [299 kB]
Get:18 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libx11-6 armhf 2:1.6.7-1+deb10u2 [689 kB]
Get:19 http://mirror.truenetwork.ru/raspbian/raspbian buster/main armhf libxml2 armhf 2.9.4+dfsg1-7+deb10u2 [572 kB]
Get:20 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-dev armhf 1:1.20210527-1 [400 kB]
Get:21 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel armhf 1:1.20210527-1 [78.9 MB]
Get:22 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-bin armhf 1:1.20210527-1 [341 kB]
Get:23 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi0 armhf 1:1.20210527-1 [847 kB]
Get:24 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-bootloader armhf 1:1.20210527-1 [4,493 kB]
Get:25 http://archive.raspberrypi.org/debian buster/main armhf linux-libc-dev armhf 1:1.20210527-1 [1,011 kB]
Get:26 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-sys-mods armhf 20210706 [12.2 kB]
Get:27 http://archive.raspberrypi.org/debian buster/main armhf pi-bluetooth all 0.1.17 [5,608 B]
Get:28 http://archive.raspberrypi.org/debian buster/main armhf python-gpiozero all 1.6.2-1 [118 kB]
Get:29 http://archive.raspberrypi.org/debian buster/main armhf python-spidev armhf 20200602~200721-1~buster [11.8 kB]
Get:30 http://archive.raspberrypi.org/debian buster/main armhf python3-gpiozero all 1.6.2-1 [121 kB]
Get:31 http://archive.raspberrypi.org/debian buster/main armhf python3-spidev armhf 20200602~200721-1~buster [11.7 kB]
Get:32 http://archive.raspberrypi.org/debian buster/main armhf rpi-eeprom armhf 12.9-1 [701 kB]
Get:33 http://archive.raspberrypi.org/debian buster/main armhf rpi-update all 20210618 [7,920 B]
Fetched 138 MB in 25s (5,589 kB/s)
Reading changelogs... Done
Extracting templates from packages: 100%
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'keyboard-configuration' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

```

XECDesign, 2021-07-09:
That looks more like a corrupted sd card. You could try `sudo apt clean && sudo apt install --reinstall keyboard-configuration', but you'll probably run into more issues at some point if the sd card or power supply isn't reliable.

lurch, 2021-07-10:
> 2021-05-07-raspios-buster-armhf-lite.img

The `armhf` indicates that it's a 32-bit Raspberry Pi OS image, not a 64-bit Raspberry Pi OS (which is what this repo is for).
For future reference, I believe issues with the 32-bit Raspberry Pi OS should be reported at https://github.com/rpi-distro/repo

closed 2021-07-12 by ghollingworth

heX16, 2021-07-13:
> That looks more like a corrupted sd card. You could try `sudo apt clean && sudo apt install --reinstall keyboard-configuration', but you'll probably run into more issues at some point if the sd card or power supply isn't reliable.

I rewrote the image and everything worked. 
I used the "verify after write" flag, but apparently this is not enough. Judging by the reviews on the internet: Using an sd-card for a 27x7 server is a very bad idea.

pelwell, 2021-07-13:
That is a really high uptime requirement.

pelwell, 2021-07-13:
Although The Beatles were in favour of 24x8.