raspberrypi/Raspberry-Pi-OS-64bit #166: gsm modem does not work after reboot or after change to wifi

melonius, 2021-07-11:
usually I work with various gsm modems on raspberry, on reboot It connect again automatically.
ModemManager

ghollingworth, 2021-07-12:
Which date versions of Raspberry Pi OS (32 bit) have you tried it on?

http://downloads.raspberrypi.org/raspios_armhf/images/

Which data versions of Raspberry Pi OS (64 bit) have you tried it on?

http://downloads.raspberrypi.org/raspios_arm64/images/

melonius, 2021-07-12:
Raspberry Pi OS (32 bit), I did noy try it. 

raspios_arm64-2020-08-24/ | 2020-08-24 18:56   but It was upgraded via apt on 2021-07-10

ghollingworth, 2021-07-12:
In that case it's possible this is a generic Debian / Linux / Raspberry Pi OS problem, not a 64bit specific problem.  We can't help you here (since we don't have the hardware), you'll have to work out which versions work and which don't, working out which particular release caused the problem so we can identify where the issue is.

Neither network-manager or modemmanager are installed by default, but it's likely they interfere with dhcpcd which is what Raspberry Pi OS uses to manage network interfaces

Gordon

melonius, 2021-07-12:
Thanks, I will try on Raspberry Pi OS (32 bit), for get more information

melonius, 2021-07-18:
I tried on Raspberry Pi OS (32 bit) and I have the same problem.
I will move the topic to the general Raspberry Pi OS forum
Thanks