raspberrypi/Raspberry-Pi-OS-64bit #168: any idea to use pi-camera on raspbian 64 bit?

Painadath, 2021-07-18:
i recently switched to 64 bit and the only problem which matter me is 64 bit doesn't support pi-camera. Because of i am using my pi for online class i am forced to use 32 bit any idea to use pi-camera on 64 bit?

JamesH65, 2021-07-18:
Currently picamera does not work on the 64 bit kernel.Please note that the 64bit distro is still in testing, it has not been officially released as  final product yet.

Painadath, 2021-07-19:
@JamesH65 I know but is there any way to do it like compiling drivers or something please

JamesH65, 2021-07-19:
If it was that easy to fix, we would have done so. This is a very difficult issue, that in all likelihood will not be fixed, but instead a new picamera will need to be developed that uses libcamera instead of the MMAL API to the firmware camera code.

We already have libcamera equivalents for raspistill etc.

Painadath, 2021-07-20:
@JamesH65 i suppose the drivers are open source then why it can't support 64bit can you please give me complete problems of it?

Painadath, 2021-07-20:
> libcamera is open source, but raspistill relies on the firmware which is closed source.

@pelwell but i need to use it for google meet will it work if i compile it?

6by9, 2021-07-20:
Do you mean the Pi camera module, or https://github.com/waveform80/picamera/ ?

The Pi camera module is supported through V4L2 and should just work with Google Meet or any other standard V4L2 app.

If you mean picamera library, then it relies on MMAL, which isn't supported on 64bit.
https://github.com/raspberrypi/userland/issues/688 is the tracking issue, but it's a low priority.

If you wish to investigate, then feel free. The IPC has a 32bit field that is used to carry a user pointer, and which is then returned with the buffer. This obviously fails when a user pointer is 64bits.
https://github.com/raspberrypi/userland/pull/586 added a simple lookup table where the 64bit pointer was stashed whilst "in transit" with the firmware, and allowed the 64bit value to be retrieved when the buffer was returned, at which point the lookup was freed.
Under certain circumstances (largely with video decode) a freed lookup value was returned, which obviously failed. It's possible that the firmware is doing something wrong and returning the same buffer/handle twice (hopefully not), or there is a race condition between alloc/lookup/free (I've looked and convinced myself it's clean, and adding mutexes doesn't solve the problem).

Painadath, 2021-07-20:
> Do you mean the Pi camera module, or https://github.com/waveform80/picamera/ ?
> 
> The Pi camera module is supported through V4L2 and should just work with Google Meet or any other standard V4L2 app.
> 
> If you mean picamera library, then it relies on MMAL, which isn't supported on 64bit.
> [raspberrypi/userland#688](https://github.com/raspberrypi/userland/issues/688) is the tracking issue, but it's a low priority.
> 
> If you wish to investigate, then feel free. The IPC has a 32bit field that is used to carry a user pointer, and which is then returned with the buffer. This obviously fails when a user pointer is 64bits.
> [raspberrypi/userland#586](https://github.com/raspberrypi/userland/pull/586) added a simple lookup table where the 64bit pointer was stashed whilst "in transit" with the firmware, and allowed the 64bit value to be retrieved when the buffer was returned, at which point the lookup was freed.
> Under certain circumstances (largely with video decode) a freed lookup value was returned, which obviously failed. It's possible that the firmware is doing something wrong and returning the same buffer/handle twice (hopefully not), or there is a race condition between alloc/lookup/free (I've looked and convinced myself it's clean, and adding mutexes doesn't solve the problem).

@6by9 i just want to use my picamera v2 with google meet for my online class in 64 bit

6by9, 2021-07-20:
Enable the camera from raspi-config, reboot, and /dev/video0 should exist. Google Meet should use that happily.

Painadath, 2021-07-20:
> Enable the camera from raspi-config, reboot, and /dev/video0 should exist. Google Meet should use that happily.

@6by9 i tried it but the google meet doesn't take it but on my 32 bit it does

6by9, 2021-07-20:
Does /dev/video0 exist?

Painadath, 2021-07-20:
> Does /dev/video0 exist?

let me check

Painadath, 2021-07-20:
@6by9 yes the video0 existes!!!

6by9, 2021-07-20:
And the other standard questions:
- Which version of Pi?
- What kernel version (```uname -a```)
- What firmware version (```vcgencmd version```)
- Is the camera detected (```vcgencmd get_camera```)

As /dev/video0 exists, install qv4l2, run it, and confirm you get images from it.
Otherwise that sounds like Google Meet (presumably via Chromium) is getting confused.
Do you get a choice of camera devices? It should be bcm2835-v4l2 that is in use, although that may be referred to as "mmal service X.X".

Painadath, 2021-07-20:
but google meet says no camera connected @6by9

Painadath, 2021-07-20:
@6by9 ^^

Painadath, 2021-07-20:
let me try

> And the other standard questions:
>
> * Which version of Pi?
> * What kernel version (`uname -a`)
> * What firmware version (`vcgencmd version`)
> * Is the camera detected (`vcgencmd get_camera`)
>
> As [...]

and also i don't get a choice on camera devices

Painadath, 2021-07-20:
> And the other standard questions:
>
> * Which version of Pi?
> * What kernel version (`uname -a`)
> * What firmware version (`vcgencmd version`)
> * Is the camera detected (`vcgencmd get_camera`)
>
> As [...]

@6by9 qv4l2 detect camera but chromium doesn't

6by9, 2021-07-20:
Sounds like a Chromium configuration issue. 
Please could you confirm the output of ```v4l2-ctl -D``` to ensure there's nothing odd there.

Painadath, 2021-07-20:
@6by9 
```
pi@raspberrypi:~ $ v4l2-ctl -D
Driver Info:
	Driver name      : bm2835 mmal
	Card type        : mmal service 16.1
	Bus info         : platform:bcm2835-v4l2-0
	Driver version   : 5.10.17
	Capabilities     : 0x85200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
pi@raspberrypi:~ $ 
```

Painadath, 2021-07-20:
@ghollingworth @JamesH65

ghollingworth, 2021-07-20:
Don't gratuitously ping people, we see the changes anyway and it's clear that 6by9 is already helping.

Painadath, 2021-07-20:
@ghollingworth sry

6by9, 2021-07-27:
This does appear to be a Chromium issue.

Use Firefox (```sudo apt install firefox-esr```) and Google Meet immediately sees the camera module and gives me a preview. I don't have a mic attached, so didn't join a meeting.

Painadath, 2021-07-27:
@6by9 It Does not work i tried it before

6by9, 2021-07-28:
I've just tried it with an HQ camera and USB sound card hooked up. Works fine using firefox-esr 78.12.0esr-1 straight out of the Debian repos.

@XECDesign Do I recall there were some configuration options that were tweaked in the 32bit Chromium builds around a year ago to get various video conferencing systems working in Chromium?
Then again we appear to be running the stock Debian Chromium builds on 64bit, so I guess changing options has quite an overhead. I also note that 64bit is still on Chromium 89.0.4389, whilst 32bit is only 88.0.4324. It's unlikely to be a regression there though.

XECDesign, 2021-07-28:
We don't have our build of Chromium in the 64-bit build yet. I'll try to get it done today and tomorrow then we can re-test and see if there are any other options that need to be tweaked.