raspberrypi/Raspberry-Pi-OS-64bit #169: Problems using a second USB connected SSD

bogflap, 2021-07-21:
I am booting from a USB attached SSD and all works well. However plugging in a second USB SSD formatted as EXT4 crashed the system, had to be power cycled to recover. Reformatted the second SSD as FAT32 and the system stayed up but the drive was marked as read only, commands like mkdir are silently ignored. I want to use the second drive as a backup.
Not to much detail here hoping that these details will prompt a "ah yes" reply.
If needed I can try this on a 32bit version but am waiting for another SSD to arrive.

OS version is 10 (buster), Pi4 Model B Rev 1.4 8GB
Boot SSD is a Samsung T5 (I think) and the second SSD is a Samsung T7(definitely).

JamesH65, 2021-07-21:
Sounds like you ran out of power to drive two devices at the same time.

lurch, 2021-07-21:
...which can normally be resolved by using a powered USB hub, rather than plugging the SSD directly into the Pi.

bogflap, 2021-07-21:
Silly me. Each SSD is marked as requiring 5V at 1.5A so two of them is very
likely to overwhelm the power supply.
Many thanks for the quick response

FibreFoX, 2021-09-03:
@bogflap ;) you might want to close this issue then, makes people more happy to see this resolved

closed 2021-09-04 by bogflap