raspberrypi/Raspberry-Pi-OS-64bit #170: all graphics are shifted right by one pixel

hereswalldo, 2021-07-21:
REPRO:  boot to the desktop, open any window, move it off the right edge... see weird column of pixels on left edge :(

I've been testing 64-bit Pi OS for several months (1920x1080 on HDMI) and noticed the left-most column of pixels looked "funny"... it turns out, the graphics *always* wrap around off the right edge of the screen, and show 1 pixel on the left edge.

This even happens when I paint the screen directly in Pygame using ARM64 ASM code. Must be some low-level GPU config bug?

Anyway, thanks to all developers, and 3 cheers for Pi OS and 64-bit ARM :)

ghollingworth, 2021-07-21:
I believe this has now been fixed in the KMS driver, so might be fixed with sudo rpi-update

If not then it should be fixed soon...

popcornmix, 2021-07-22:
Yes, rpi-update kernel contains the fix for this. It will appear in an apt update soon.

hereswalldo, 2021-07-30:
hey...  it's fixed!  :)

closed 2021-07-30 by pelwell