raspberrypi/Raspberry-Pi-OS-64bit #171: Won't Boot with Composite Output

evansste12432, 2021-07-24:
I configured my Raspberry Pi 4 to boot with the composite output.  However, when I try to boot with the 64-Bit Raspberry Pi OS, it appears to freeze on the "Welcome to the Raspberry Pi Desktop" page.

I configured my output to the 3.5 mm composite jack by altering the `config.txt` file.  I placed the following line in the file:

`enable_tvout=1`

Doing this causes the Pi to seemingly freeze instead of boot up.

I recognize that, for the 64-bit OS, the Pi is expected to run slower with the composite output.  However, this appears to be freezing, since it doesn't get beyond this point, when booting up.

popcornmix, 2021-07-24:
Can you post complete contents of config.txt and cmdline.txt and output of:
```
vcgencmd version
uname -a
```
How do you determine it has frozen? Just lack of display output? Lack of ssh ability? Lack of uart output?

evansste12432, 2021-07-24:
I just tried it with a different TV, and it booted up with no problems.  It would be odd that the TV is the culprit, since it was showing everything just fine until it hit that "welcome" page.  Nevertheless, it  works fine now.

Before trying it on a different TV, I pulled out the SD card, in order to view the contents of the `config.txt` and `cmdline.txt` files.  Perhaps the card wasn't seated very well, and was seated more properly the second time I put it in.  In any case, this issue doesn't seem to be an issue anymore.

Thanks for your time, attention, and for responding so quickly.

closed 2021-07-24 by pelwell

6by9, 2021-07-24:
Isn't the 64 bit OS using vc4-kms-v3d? That only gained support for composite out in the last couple of weeks, and gained support for PAL output at that same point (previously it was pretty much NTSC only). Prior to that the timings were slightly off as well, so it would depend on how tolerant the display was to non-standard timings.

evansste12432, 2021-07-24:
@6by9, that could be a possible explanation.  However, the image that I downloaded, has a date of May 7th, 2021.  The exact image file name is `2021-05-07-raspios-buster-arm64.img`.  Or perhaps the date reads July 5th, 2021.