raspberrypi/Raspberry-Pi-OS-64bit #172: Widevine Doesn't Seem to Work

evansste12432, 2021-07-25:
After installing 64-Bit Raspberry Pi OS, and then installing Widevine, I've noticed that Widevine doesn't show up in the Chromium browser.

After a fresh install of 64-Bit Raspberry Pi OS, I did the following, in order to install Widevine:

`sudo apt update`
`sudo apt full-upgrade`
`sudo apt install libwidevinecdm0`

After running these commands, Widevine doesn't show up as being installed in Chromium.  If I open the Chromium web browser, and type `chrome://components`, Widevine isn't listed.

According to the following article:

https://pimylifeup.com/raspberry-pi-widevine/comment-page-1/#comment-62789

and also according to a video, which I watched on YouTube, it's supposed to be as easy as simply running the above three commands in order to install Widevine.  Has something changed, or are these sources simply wrong about how to install Widevine?

Thanks for your time and attention.

lurch, 2021-07-25:
See also https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/search?q=widevine&type=issues

evansste12432, 2021-07-28:
Thanks for the link, @lurch.  The craziest thing is that I actually had this working, less than a week ago.  I installed the 64-bit Raspberry Pi OS, and tried playing an amazon video.  It worked, but I was watching it via VNC, and the video was a bit choppy.  I overwrote the SD card, not thinking that it'd be difficult to set it up again.  However, now, after reinstalling the 64-bit Raspberry Pi OS, it just doesn't work.

Unfortunately, I have no way of showing what I did, the first time.  All I know is that it was working.  I was watching the beginning of "Invincible" on Amazon Prime, on my Raspberry Pi 4, via a VNC connection.  I really wish I had backed up that SD card before writing over its contents.

lurch, 2021-07-28:
Did you mean [VNC](https://www.realvnc.com/en/), or [VLC](https://www.videolan.org/)? Similar acronyms, very different programs :wink:

evansste12432, 2021-07-28:
VNC.  I have, both, a Raspberry Pi 4 and a Raspberry Pi 3.  I currently don't have my HDMI cable for my Raspberry Pi 4.  So I installed 64-Bit Raspberry Pi OS on my Raspberry Pi 3, and then noticed that it was extremely slow.  So I popped the SD card into my Raspberry Pi 4, and used tightVNCserver to view it via my laptop.  That's when I was able to see it working.  I've tried to do this again, to no avail.  I may try it again, because it was working.

lurch, 2021-07-28:
AFAIK VNC has nothing to do with widevine?? :confused:

evansste12432, 2021-07-28:
I don't think VNC has anything to do with widevine, either.  I was just explaining the specific means, by which, it ended up working.

lurch, 2021-07-28:
Ah, sorry I misunderstood you.

XECDesign, 2021-08-31:
> The craziest thing is that I actually had this working, less than a week ago.

If you can reproduce that, then I can take a look. To the best of my knowledge, that's no possible without having a 32-bit version of chromium installed (which is a solution we may go with).

evansste12432, 2021-09-19:
I just got Netflix to work on my Raspberry Pi 4.  I'm using the 32-bit version of the Raspberry Pi OS.

I'm beginning to think that I popped in an SD card that already had the 32-bit version of the Raspberry Pi OS, and noticed that it could play videos in Amazon Prime.

Once Google dropped Widevine support for 32-bit machines, I figured I'd see if it worked on the Raspberry Pi 4, which uses 64-bit hardware.  So I probably popped in the first SD card which already had the Raspberry Pi OS on it, and noticed that it worked.

I'm glad that this works.  I don't really care that it's 32-bit, as long as it works.

So, from what I can tell, Widevine doesn't care whether or not your operating system is 64-bit.  It only cares that your hardware is 64-bit.  That's the way it seems, anyway, since I can't get any streaming platforms, other than YouTube, to work on my laptop; which has 32-bit hardware.

sq6sfo, 2022-05-11:
I see it's a quite old issue, but I've managed to find a workaround just recently.
It's true that libwidevinecdm0 seems to be provided only in 32-bit version and does not work with 64-bit chromium.
If you still want to watch widevine protected stuff while keeping 64-bit os it's possible to use dpkg multiarch functionality to install 32-bit version of the browser only:
`sudo dpkg --add-architecture armhf` <- this will add 32-bit arm architecture
`sudo apt update`
`sudo apt install chromium-browser:armhf` <- this will remove chromium-browser and install its 32-bit version along with dependencies
I can confirm it works flawlessly now with recent RpiOS-64  (tested with Netflix and HBO)

Botspot, 2022-07-17:
We developers at [Pi-Apps](https://github.com/Botspot/pi-apps) are watching the Widevine space closely to see if Google plans to enable Widevine in its 64-bit ChromeOS userspace.
[This commit](https://chromium.googlesource.com/chromium/src.git/+/2868ab1359c64c5e063f259d46bd655bd8ad8bf2) to the Chromium source code was made in March, so they appear to be heading that way internally.

satyamvatsal, 2022-11-06:
> 

I done the same thing but Spotify is not playing even then
but it shows 32 bit chromium

satyamvatsal, 2022-11-06:
> 

Is it working now?

popcornmix, 2022-11-07:
widevine only supports 32-bit arm.
widevine works on RPiOS 32 chromium by default.

On 64-bit RPiOS you can install 32-bit chromium and widevine and it will work.
See [this blog post](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/)

XECDesign, 2023-02-20:
A 64bit widevine build has just been added and chromium-browser updated accordingly.

popcornmix, 2023-02-20:
libgcc_hide.so is created from this script:
```
#!/bin/bash
OLD_ARCHIVE=${OLD_ARCHIVE:-/lib/gcc/aarch64-linux-gnu/10/libgcc.a}
NEW_ARCHIVE=${NEW_ARCHIVE:-libgcc_hide.a}
NEW_ARCHIVE_SO=${NEW_ARCHIVE_SO:-libgcc_hide.so}

OBJS="ldadd_4_4.o swp_4_4.o"
for i in $OBJS; do
 ar xv /lib/gcc/aarch64-linux-gnu/10/libgcc.a $i
 unhide.py $i
done

ar rcs $NEW_ARCHIVE $OBJS
rm $OBJS

gcc -o $NEW_ARCHIVE_SO -shared -Wl,--whole-archive $NEW_ARCHIVE -Wl,--no-whole-archive
```
using unhide.py from [here](https://stackoverflow.com/questions/54664759/keep-all-exported-symbols-when-creating-a-shared-library-from-a-static-library)

tmm1, 2023-02-20:
You could also compile it yourself using the code here:  https://github.com/xbmc/inputstream.adaptive/pull/1130/files#diff-cb49b4f5a6e2a38d51a696be771430404f8a6c7f7012b92d5d27da461825f790R1674

theofficialgman, 2023-02-21:
maybe someone in this thread can be of assistance

how could the widevine lib be patched to not require `fcntl64` which was added in glibc 2.28 and is not available on ubuntu bionic?

XECDesign, 2023-02-21:
> You could also compile it yourself using the code here: https://github.com/xbmc/inputstream.adaptive/pull/1130/files#diff-cb49b4f5a6e2a38d51a696be771430404f8a6c7f7012b92d5d27da461825f790R1674
> 
> ```c
>  __attribute__((target("no-outline-atomics")))
>  int32_t __aarch64_ldadd4_acq_rel(int32_t value, int32_t *ptr)
>  {
>    return __atomic_fetch_add(ptr, value, __ATOMIC_ACQ_REL);
>  }
> 
>  __attribute__((target("no-outline-atomics")))
>  int32_t __aarch64_swp4_acq_rel(int32_t value, int32_t *ptr)
>  {
>    return __atomic_exchange_n(ptr, value, __ATOMIC_ACQ_REL);
>  }
> ```

I've just tried this approach and it also works. I'll switch to it, since it makes GPL compliance simpler.

theofficialgman, 2023-02-24:
I am not sure which user is Serge Schneider <serge@raspberrypi.com> as the debian maintainer of the glibc package.

I am attempting to modify the glibc 2.31 patchs for glibc 2.27 (bionic) so that I can use widevine.
I have modified them, then apply, and the deb builds successfully but I get this error when running chromium or with ldd on the widevine binary.

```
Inconsistency detected by ld.so: get-dynamic-info.h: 143: elf_get_dynamic_info: Assertion `info[DT_RELRENT]->d_un.d_val == sizeof (ElfW(Relr))' failed!
```
the patches themselves that were added to ubuntu bionic glibc are here:
[glibc-2.27-patches.zip](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/10819734/glibc-2.27-patches.zip)

theofficialgman, 2023-02-24:
hid my preview comment (the patch had a mistake)

was able to get widevine working on ubuntu bionic

**GLIBC PATCHES**
GLIBC 2.36+:
<https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/devel/glibc/patches/arm/glibc-HACK-Don-t-check-GLIBC_ABI_DT_RELR-support-for-Chrom.patch>
<https://cdn.discordapp.com/attachments/567232809475768320/1078517350137090098/glibc-tls-libwidevinecdm.so-since-4.10.2252.0-has-TLS-with.patch>

GLIBC 2.35 and lower:
<https://cdn.discordapp.com/attachments/567232809475768320/1078517349776359485/glibc-2.27-sht_relr_new.diff>
<https://cdn.discordapp.com/attachments/567232809475768320/1078517350137090098/glibc-tls-libwidevinecdm.so-since-4.10.2252.0-has-TLS-with.patch>

**symbols/functions for widevine to LD_PRELOAD**
GLIBC 2.27:
```
#include <fcntl.h>
#include <stdarg.h>
#include <stdint.h>

int fcntl64(int fd, int cmd, void*arg) {
  return (fcntl(fd, cmd, arg));
}

__attribute__((target("no-outline-atomics")))
int32_t __aarch64_ldadd4_acq_rel(int32_t value, int32_t *ptr)
{
return __atomic_fetch_add(ptr, value, __ATOMIC_ACQ_REL);
}

__attribute__((target("no-outline-atomics")))
int32_t __aarch64_swp4_acq_rel(int32_t value, int32_t *ptr)
{
return __atomic_exchange_n(ptr, value, __ATOMIC_ACQ_REL);
}
```
GLIBC 2.28+
```
#include <fcntl.h>
#include <stdarg.h>
#include <stdint.h>

__attribute__((target("no-outline-atomics")))
int32_t __aarch64_ldadd4_acq_rel(int32_t value, int32_t *ptr)
{
return __atomic_fetch_add(ptr, value, __ATOMIC_ACQ_REL);
}

__attribute__((target("no-outline-atomics")))
int32_t __aarch64_swp4_acq_rel(int32_t value, int32_t *ptr)
{
return __atomic_exchange_n(ptr, value, __ATOMIC_ACQ_REL);
}
```

for GLIBC 2.27 and 2.28, need to mark the required symbols as WEAK by patching the widevine binary
follow this guide: https://www.lightofdawn.org/wiki/wiki.cgi/NewAppsOnOldGlibc

matthuisman, 2023-02-25:
@XECDesign just to be clear, there is now a new build of chrome on 64bit rpi os with patches needed for arm64 bit widevine cdm? And I suspect rpi are now hosting the 64bit widevine cdm as well now?