raspberrypi/Raspberry-Pi-OS-64bit #174: 64-bit raspberry pi issue with libstdc++ package

vs1612, 2021-07-28:
I am trying to run my code on raspberry pi4, i created a conda environment . whenver i am runnning my code i am getting import error as follows.
Traceback (most recent call last):
  File "main1.py", line 6, in <module>
    from data_loader import organiser
  File "/home/pi/Deep_Audio_Net/data_loader/organiser.py", line 6, in <module>
    from exp_run import audio_feature_extractor as afe
  File "/home/pi/Deep_Audio_Net/exp_run/audio_feature_extractor.py", line 2, in <module>
    import librosa
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/__init__.py", line 211, in <module>
    from . import core
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/core/__init__.py", line 5, in <module>
    from .convert import *  # pylint: disable=wildcard-import
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/core/convert.py", line 7, in <module>
    from . import notation
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/core/notation.py", line 8, in <module>
    from ..util.exceptions import ParameterError
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/util/__init__.py", line 87, in <module>
    from ._nnls import *  # pylint: disable=wildcard-import
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/librosa/util/_nnls.py", line 13, in <module>
    import scipy.optimize
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/scipy/optimize/__init__.py", line 400, in <module>
    from .optimize import *
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/scipy/optimize/optimize.py", line 36, in <module>
    from ._numdiff import approx_derivative
  File "/home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/scipy/optimize/_numdiff.py", line 8, in <module>
    from ._group_columns import group_dense, group_sparse
ImportError: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/pi/archiconda3/envs/daic/lib/python3.7/site-packages/scipy/optimize/_group_columns.cpython-37m-aarch64-linux-gnu.so)

could some please help me with this.

lurch, 2021-07-28:
As I mentioned in your other issue, it sounds like this is something that you need to ask the conda developers about?

vs1612, 2021-07-28:
Ok , Thank you.

Regards,
Vikas

closed 2021-07-28 by vs1612