raspberrypi/Raspberry-Pi-OS-64bit #176: sudo apt update && sudo apt -y upgrade gives error for bluetooth service and kills the image

teq0, 2021-08-04:
After booting on RPi 3 B+ run 

`sudo apt update && sudo apt -y upgrade`

Amongst all the updates there's an error:

```Job for hciuart.service failed because the control process exited with error code.
See "systemctl status hciuart.service" and "journalctl -xe" for details.

pi@raspberrypi:~ $ systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2021-08-03 01:31:42 BST; 21min ago

Aug 03 01:31:12 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 03 01:31:42 raspberrypi btuart[4355]: Initialization timed out.
Aug 03 01:31:42 raspberrypi btuart[4355]: bcm43xx_init
Aug 03 01:31:42 raspberrypi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Aug 03 01:31:42 raspberrypi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Aug 03 01:31:42 raspberrypi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
pi@raspberrypi:~ $ journalctl -u hciuart.service
-- Logs begin at Fri 2021-05-07 17:11:25 BST, end at Tue 2021-08-03 01:39:36 BST. --
May 07 17:11:30 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
May 07 17:11:39 raspberrypi btuart[382]: bcm43xx_init
May 07 17:11:39 raspberrypi btuart[382]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
May 07 17:11:39 raspberrypi btuart[382]: Set BDADDR UART: b8:27:eb:98:8d:f0
May 07 17:11:39 raspberrypi btuart[382]: Set Controller UART speed to 3000000 bit/s
May 07 17:11:39 raspberrypi btuart[382]: Device setup complete
May 07 17:11:39 raspberrypi systemd[1]: Started Configure Bluetooth Modems connected by UART.
Aug 03 01:30:39 raspberrypi systemd[1]: Stopping Configure Bluetooth Modems connected by UART...
Aug 03 01:30:41 raspberrypi systemd[1]: hciuart.service: Succeeded.
Aug 03 01:30:41 raspberrypi systemd[1]: Stopped Configure Bluetooth Modems connected by UART.
Aug 03 01:30:41 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 03 01:31:11 raspberrypi btuart[4259]: Initialization timed out.
Aug 03 01:31:11 raspberrypi btuart[4259]: bcm43xx_init
Aug 03 01:31:11 raspberrypi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Aug 03 01:31:11 raspberrypi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Aug 03 01:31:11 raspberrypi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Aug 03 01:31:12 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 03 01:31:42 raspberrypi btuart[4355]: Initialization timed out.
Aug 03 01:31:42 raspberrypi btuart[4355]: bcm43xx_init
Aug 03 01:31:42 raspberrypi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Aug 03 01:31:42 raspberrypi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Aug 03 01:31:42 raspberrypi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
```

After this the image no longer boots. My monitor is broken atm so I can't see where it's failing. Reproduced using two different SD cards on two different RPis.

nkkfs, 2021-09-05:
+1 on RPi 4B. 
Cause of this probably in Home Assistant cannot connect to BLE Sensors.

pelwell, 2021-09-06:
What does `apt info pi-bluetooth bluez bluez-firmware` report? I'm on version 0.1.17 of pi-bluetooth, 5.50-1.2~deb10u1+rpt1 of bluez and 1.2-4+rpt8 of  bluez-firmware, and it's working fine.

teq0, 2021-09-06:
After `apt upgrade` it doesn't even boot, so I can't tell you. Before that it's whatever is in the base distro.

pelwell, 2021-09-07:
You were able to run `systemctl status hciuart.service`, so that would have been the time to look at versions (not that I'm asking you to be clairvoyant).

I ran the update and didn't see a problem. WIthout knowing the version numbers you upgraded to I don't know how to proceed.

teq0, 2021-09-07:
OK, I'll try it again and note the versions before rebooting.

pelwell, 2021-09-08:
There is a problem with the upgrade in that it tries to restart the Bluetooth services. It's actually necessary to reboot to restart Bluetooth, and a future update will disable the automatic restart. However, this absolutely shouldn't render the device unbootable.

teq0, 2021-09-08:
Wow, 3rd time's the charm. Started from scratch, got all the same errors but this time it actually booted again after the update. I'm might try it one more time, but happy to close this. This was with yet another different SD card and a different (third) Pi, I'll go back and try the other ones again, it might have been a hardware problem. The previous SD cards weren't new. 

Out of interest, how can I check the version of the hciuart service? I googled and played around with apt and dpkg and ls mod etc but I wasn't sure what I was looking for.

pelwell, 2021-09-09:
I'm happy things are starting to work for you. That an upgrade can lead to a boot failure is alarming.

The hciuart service is installed by the pi-bluetooth package:
```
pi@raspberrypi:~$ dpkg -S /lib/systemd/system/hciuart.service
pi-bluetooth: /lib/systemd/system/hciuart.service
pi@raspberrypi:~$ apt info pi-bluetooth
Package: pi-bluetooth
Version: 0.1.15
Priority: optional
...
```

teq0, 2021-09-10:
I've now tried to reproduce this several times and it's been fine (other than the restart hciuart error), no idea what the root cause was but it's not happening anymore. Sorry if I wasted anyone's time (I definitely reproduced it before I opened the issue though).

closed 2021-09-10 by teq0