raspberrypi/Raspberry-Pi-OS-64bit #177: no sound or distorted sound with Hifiberry (DAC2 Pro)

smutbert, 2021-08-04:
Raspbian Pi OS aarch64 works great for me, except one issue. I hope this is the right place to report it:

The Hifiberry DAC2 Pro is detected correctly and in Alsa everything looks right
`$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 [HiFiBerry DAC+ HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`
but there is either nor sound, just noise or distorted sound (too low and slow, as if audio was played with a too high sample rate, but independent of the sample rate of the audio file or the mpd output).

This happens at least with aplay and mpd, Options in /boot/config.txt like
`dtoverlay=hifiberry-dacplus,slave=no
force_eeprom_read=0`
don't help.

I am running Raspbian Pi OS "buster":
`$ uname -a
Linux wip 5.10.52-v8+ #1440 SMP PREEMPT Tue Jul 27 09:56:40 BST 2021 aarch64 GNU/Linu`

popcornmix, 2021-08-05:
It would be useful if you could confirm if this is an issue with RPiOS 64bit, or a general issue.
Install 32bit RPiOS to a spare sdcard and check if it has the same issue. If it does then https://github.com/raspberrypi/linux/issues would be a better place for bug report.

smutbert, 2021-08-05:
I switched to the 32bit version because it was not working with the 64bit variant:
```
Linux wipp 5.10.52-v7+ #1440 SMP Tue Jul 27 09:54:13 BST 2021 armv7l GNU/Linux
```
The configuration is the same and it is running fine.

pelwell, 2021-08-05:
Can you try with "arm_64bit=1" in config.txt on the 32-bit OS? That will help to locate the problem in the kernel or the userland code.

smutbert, 2021-08-05:
So then the 5.10.52-v8+ kernel will get booted the next time, right?

(I would like to know first, because I need to build a initramfs for the kernel. My /-filesystem is btrfs. It is a headless audio player and if something goes wrong, I'd need to open the case and take out the sdcard, which is quite a hassle and I already had to do it too often, when I switched from 32 to 64 bit.)

smutbert, 2021-08-05:
Ok, I tried it anyway and the problem seems to be in the userspace. Now kernel 5.10.52-v8+ is running and it still working.