raspberrypi/Raspberry-Pi-OS-64bit #178: Raspberry Pi 4/400 network interface switching between down and up status

julben, 2021-08-15:
## Current behavior:
- The interface (`eth0`) of the integrated NIC of the SoC of my Rasperry Pi 4 (and 400) is constantly switching between "up" and "down" status, as reported by `dmesg`:
```
[13311.171718] bcmgenet fd580000.ethernet eth0: Link is Down
[13313.219840] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[13393.092824] bcmgenet fd580000.ethernet eth0: Link is Down
[13396.164840] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[13432.005229] bcmgenet fd580000.ethernet eth0: Link is Down
[13435.077479] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[13562.054861] bcmgenet fd580000.ethernet eth0: Link is Down
[13565.126956] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[13639.879848] bcmgenet fd580000.ethernet eth0: Link is Down
[13642.952052] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[13851.850591] bcmgenet fd580000.ethernet eth0: Link is Down
[13854.922580] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[14254.287566] bcmgenet fd580000.ethernet eth0: Link is Down
[14257.359734] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[14309.584214] bcmgenet fd580000.ethernet eth0: Link is Down
[14312.656314] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```
- This can be seen by an ssh session freezing for a while for example.

## Expected behavior:
- `eth0` interface stays "up" after boot
- the kernel only needs to brings up the interface once (at boot)

## Steps to reproduce:
1) Flash an SD card with Raspberry Pi OS (Lite in my case) 64bits
2) Create an empty `ssh` file in the `/boot/` directory/partition
3) Insert it in a Raspberry Pi 4 or 400
4) Power up the Raspberry Pi
5) Connect via ssh to the Pi
6) Wait for the problem to occur
7) Display the kernel buffer

## Context
- This issue occurs with any fresh install of Raspberry Pi OS 64bits (I even tried with older images I still had).
- Every image was verified before flashing.
- Before reporting this issue, the system had been updated to the current versions of RPOS.
- I tried to change everything - from the power supply to the Ethernet cable, SD card, switch and even other versions of Raspberry Pi 4(00).
- Apart from the SD card, the Ethernet cable and the PSU nothing is connected to the Pi used to reproduce this issue.
- The Raspberry Pi is, of course, left in autonegociation and is assigned an IP by a DHCP server, no static IP is assigned on the device itself, not a single setting was changed.

Here are logs from a Raspberry Pi 400:

I did not find any duplicate showing this exact issue, please let me know if this is a duplicate or if I forgot any relevant detail.

lurch, 2021-08-16:
Possibly a similar issue to #132 or one of the other issues linked to from there? I wonder if it might be a problem related to your network switch or DHCP server? :man_shrugging:  Can you provide more details about the rest of your network setup?

julben, 2021-08-16:
> Possibly a similar issue to #132 or one of the other issues linked to from there? I wonder if it might be a problem related to your network switch or DHCP server? man_shrugging Can you provide more details about the rest of your network setup?

1)Like I said, I tried various solutions for everything, including DHCP servers and switches, so, no. 
2) I search for duplicates before posting. #132 does not seem to apply to this issue.

lyuboasenov, 2022-01-24:
I am having the same issue. After working correctly for some time - about 4-5 hours - the network interface goes down for no apparent reason.

I tried various fixes/workaround: 
- changing the cable
- changing the switch 
- disabling the EEE on the interface as stated on the pi forum [https://forums.raspberrypi.com/viewtopic.php?t=305820]
to no avail.

The way I was able to work around the issue is by disabling the WiFi and BT interfaces as suggested here [https://www.reddit.com/r/pihole/comments/m0myri/ethernet_eth0_link_is_down_occasionally_pihole/]. I acknowledge that this is far from a perfect solution, but I don't need the WiFi and the BT.

I haven't tried disabling either one of them and leaving the other running which may point out the culprit.

Tuinslak, 2022-09-01:
I have the exact same issue with a RPi4 that's been booted up for the first time in a few months. 

I tried both with a brand new (64bit) image using the RPi image flasher thingie, as well as a SD card that I took out of a 100% working RPi1. The issue starts to appear during Linux boot (LEDs keep going on/off). Before boot (when in the bootloader/netboot options) the ethernet seems to be behaving fine (at least in terms of LEDs and lighting up correctly).

```
[  176.235337] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  177.275273] bcmgenet fd580000.ethernet eth0: Link is Down
[  180.395396] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  181.435244] bcmgenet fd580000.ethernet eth0: Link is Down
[  184.555329] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  185.595245] bcmgenet fd580000.ethernet eth0: Link is Down
[  188.725355] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  189.755247] bcmgenet fd580000.ethernet eth0: Link is Down
[  218.875430] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  219.916636] bcmgenet fd580000.ethernet eth0: Link is Down
[  227.205495] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  228.235329] bcmgenet fd580000.ethernet eth0: Link is Down
[  240.715446] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  241.755354] bcmgenet fd580000.ethernet eth0: Link is Down
[  250.084023] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  251.125392] bcmgenet fd580000.ethernet eth0: Link is Down
[  269.835607] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  270.875625] bcmgenet fd580000.ethernet eth0: Link is Down
[  286.475594] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  288.555455] bcmgenet fd580000.ethernet eth0: Link is Down
[  291.675554] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  292.715514] bcmgenet fd580000.ethernet eth0: Link is Down
[  295.845564] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  296.875508] bcmgenet fd580000.ethernet eth0: Link is Down
[  299.995564] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  301.035497] bcmgenet fd580000.ethernet eth0: Link is Down
[  308.315604] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  309.355531] bcmgenet fd580000.ethernet eth0: Link is Down
```
And tons more of these.

There's currently two USB disks plugged in.

```
wol ~ # lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 18a5:0216 Verbatim, Ltd External Hard Drive
Bus 001 Device 004: ID 04e8:5f07 Samsung Electronics Co., Ltd Samsung STORY Station
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
wol ~ # uname -a
Linux wol 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
wol ~ # free -m
               total        used        free      shared  buff/cache   available
Mem:            3888         111        1279           3        2497        3620
Swap:              0           0           0
wol ~ # cat /etc/debian_version
11.4
wol ~ # lscpu
Architecture:                    armv7l
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
CPU max MHz:                     900.0000
CPU min MHz:                     600.0000
BogoMIPS:                        162.00
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Mmio stale data:   Not affected
Vulnerability Retbleed:          Not affected
Vulnerability Spec store bypass: Not affected
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Vulnerable
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
```

zetxx, 2022-11-17:
+1
arch arm here
connected directly to mikrotik

strange-v, 2022-12-04:
The same thing. But in my case, it stops working at all after some time.
I'm running a few containers for home automation on Pi 4, and everything was fine before switching to the x64 version (maybe the network interface was restarting, but it never hung like now).

Restarting the network via `service networking restart` doesn't help, only reboot.
```
[118778.698661] bcmgenet fd580000.ethernet eth0: Link is Down
[118785.866818] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[118809.419149] bcmgenet fd580000.ethernet eth0: Link is Down
[118816.587203] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[119657.298910] bcmgenet fd580000.ethernet eth0: Link is Down
[119664.467043] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[119699.283454] bcmgenet fd580000.ethernet eth0: Link is Down
[119705.427496] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[580153.818351] br-eb4a83954af6: port 1(veth76393ae) entered disabled state
[580153.818934] veth65be048: renamed from eth0
[580153.918194] br-eb4a83954af6: port 1(veth76393ae) entered disabled state
[580153.922665] device veth76393ae left promiscuous mode
[580153.922712] br-eb4a83954af6: port 1(veth76393ae) entered disabled state
[580154.010359] br-eb4a83954af6: port 2(vethc110d03) entered disabled state
[580154.011352] vethde1e6f5: renamed from eth0
[580154.153682] br-eb4a83954af6: port 2(vethc110d03) entered disabled state
[580154.157433] device vethc110d03 left promiscuous mode
[580154.157472] br-eb4a83954af6: port 2(vethc110d03) entered disabled state
[580154.164006] br-eb4a83954af6: port 5(veth2a22b0f) entered disabled state
[580154.164516] vethbcb0e04: renamed from eth0
[580154.233984] br-eb4a83954af6: port 5(veth2a22b0f) entered disabled state
[580154.237043] device veth2a22b0f left promiscuous mode
[580154.237065] br-eb4a83954af6: port 5(veth2a22b0f) entered disabled state
[580154.413105] br-eb4a83954af6: port 3(vetha8232f2) entered disabled state
[580154.413460] veth8609fbf: renamed from eth0
[580154.492525] br-eb4a83954af6: port 3(vetha8232f2) entered disabled state
[580154.498099] device vetha8232f2 left promiscuous mode
[580154.498155] br-eb4a83954af6: port 3(vetha8232f2) entered disabled state
[580154.691190] br-eb4a83954af6: port 6(veth2fbc92e) entered disabled state
[580154.691636] vethb1717e0: renamed from eth0
[580154.758944] br-eb4a83954af6: port 6(veth2fbc92e) entered disabled state
[580154.761555] device veth2fbc92e left promiscuous mode
[580154.761585] br-eb4a83954af6: port 6(veth2fbc92e) entered disabled state
[580158.002050] br-eb4a83954af6: port 4(veth3c987a2) entered disabled state
[580158.002807] vetha323896: renamed from eth0
[580158.059506] br-eb4a83954af6: port 4(veth3c987a2) entered disabled state
[580158.063558] device veth3c987a2 left promiscuous mode
[580158.063581] br-eb4a83954af6: port 4(veth3c987a2) entered disabled state
[580164.161607] br-eb4a83954af6: port 1(veth40f43af) entered blocking state
[580164.161629] br-eb4a83954af6: port 1(veth40f43af) entered disabled state
[580164.162529] device veth40f43af entered promiscuous mode
[580164.199133] br-eb4a83954af6: port 2(vethe7834c0) entered blocking state
[580164.199157] br-eb4a83954af6: port 2(vethe7834c0) entered disabled state
[580164.199436] device vethe7834c0 entered promiscuous mode
[580164.199723] br-eb4a83954af6: port 2(vethe7834c0) entered blocking state
[580164.199740] br-eb4a83954af6: port 2(vethe7834c0) entered forwarding state
[580164.203386] br-eb4a83954af6: port 2(vethe7834c0) entered disabled state
[580164.265884] IPv6: ADDRCONF(NETDEV_CHANGE): veth5ce2d08: link becomes ready
[580164.266038] IPv6: ADDRCONF(NETDEV_CHANGE): veth40f43af: link becomes ready
[580164.266124] br-eb4a83954af6: port 1(veth40f43af) entered blocking state
[580164.266136] br-eb4a83954af6: port 1(veth40f43af) entered forwarding state
[580164.405309] br-eb4a83954af6: port 3(veth9a6aa64) entered blocking state
[580164.405337] br-eb4a83954af6: port 3(veth9a6aa64) entered disabled state
[580164.417056] device veth9a6aa64 entered promiscuous mode
[580164.424169] br-eb4a83954af6: port 3(veth9a6aa64) entered blocking state
[580164.424197] br-eb4a83954af6: port 3(veth9a6aa64) entered forwarding state
[580164.436564] br-eb4a83954af6: port 4(veth586f3e8) entered blocking state
[580164.436589] br-eb4a83954af6: port 4(veth586f3e8) entered disabled state
[580164.436767] device veth586f3e8 entered promiscuous mode
[580164.437012] br-eb4a83954af6: port 4(veth586f3e8) entered blocking state
[580164.437024] br-eb4a83954af6: port 4(veth586f3e8) entered forwarding state
[580164.475929] br-eb4a83954af6: port 3(veth9a6aa64) entered disabled state
[580164.476125] br-eb4a83954af6: port 4(veth586f3e8) entered disabled state
[580164.476201] IPv6: ADDRCONF(NETDEV_CHANGE): vethe7834c0: link becomes ready
[580164.476303] br-eb4a83954af6: port 2(vethe7834c0) entered blocking state
[... 114 lines cut ...]
[580182.627475] br-eb4a83954af6: port 1(veth3477e71) entered blocking state
[580182.627487] br-eb4a83954af6: port 1(veth3477e71) entered forwarding state
[580183.319823] eth0: renamed from veth162d98e
[580183.383151] IPv6: ADDRCONF(NETDEV_CHANGE): veth7d69017: link becomes ready
[580183.383281] br-eb4a83954af6: port 2(veth7d69017) entered blocking state
[580183.383293] br-eb4a83954af6: port 2(veth7d69017) entered forwarding state
[580183.415362] eth0: renamed from vethe887e6f
[... cut ...]
[580183.611023] br-eb4a83954af6: port 6(veth4285f7c) entered blocking state
[580183.611035] br-eb4a83954af6: port 6(veth4285f7c) entered forwarding state
[580183.611652] IPv6: ADDRCONF(NETDEV_CHANGE): veth8c21765: link becomes ready
[580183.611754] br-eb4a83954af6: port 5(veth8c21765) entered blocking state
[580183.611773] br-eb4a83954af6: port 5(veth8c21765) entered forwarding state
[632797.385528] bcmgenet fd580000.ethernet eth0: Link is Down
[632803.529586] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

hyptx, 2023-01-17:
Was there any decent resolution. I am running into this. Can I just disable the eth0? I only use wifi

pelwell, 2023-01-18:
A relatively recent change (mid-December) makes `dtparam=eee=off` (disabling Energy Efficient Ethernet support) work on Pi 4/400. `sudo apt update` and `sudo apt install raspberrypi-kernel`, and add `dtparam=eee=off` to config.txt.

HackerDaGreat57, 2023-01-23:
Also happening in Ubuntu 22.04 LTS x64 fully updated

acavalin, 2023-01-26:
Hello everyone, using `dtparam=eee=off` I can confirm a stable connection in the last 7 days without a single hiccup! 💯 

> A relatively recent change (mid-December) makes `dtparam=eee=off` (disabling Energy Efficient Ethernet support) work on Pi 4/400. `sudo apt update` and `sudo apt install raspberrypi-kernel`, and add `dtparam=eee=off` to config.txt.

chris-deluca, 2023-01-31:
I posted this same info on a similar thread.  

I adjusted the proposed setting and have started to see some more stable behavior.  The attached chart shows the network periodically going up and down - I have 2 different forms of monitoring one internal and one external - while the "internal" chart collection process (shown) fails periodically (see dips), the "external" collection process (not provided) shows no such drop-offs.

The sites being monitored are 3 separate servers hosting 8 different resources. All resources are docker containers. The ONLY change made was to ssh into my pi (running Ubuntu 22.04.1) and execute the following command: ethtool --set-eee eth0 eee off - I didn't bounce containers or do any type of resetting. Also interesting is the "scrape time" shown in the 3rd chart reduced from almost 20 seconds to near 0.  This chart is now for data covering more than 2 days.

[image]

jvcsw, 2023-02-14:
Hi, 

do you need to run this command again on every reboot?

Thanks

lurch, 2023-02-14:
> do you need to run this command again on every reboot?

Which "command" are you talking about? `config.txt` is a file on the `/boot` partition, which the Pi reads every time it boots.

jvcsw, 2023-02-14:
> > do you need to run this command again on every reboot?
> 
> Which "command" are you talking about? `config.txt` is a file on the `/boot` partition, which the Pi reads every time it boots.

...following command: ethtool --set-eee eth0 eee off

lurch, 2023-02-14:
Right. So AIUI you can run that ethtool command and it'll have an immediate effect (but not persist over a reboot), or you can add the dtparam to config.txt which will only activate after a reboot (but then apply after every reboot).