raspberrypi/Raspberry-Pi-OS-64bit #179: can not mount xfs filesystem on xfs-partition

x2x3y, 2021-08-25:
I wanted to set up a samba cluster with glusterfs. Unfortunately I didn't succeed in the 64Bit OS.
All installation descriptions on the internet state that glusterFS requires an XFS formatted file system.
Creating an xfs formatted xfs partition worked well with the previously installed xfsprogs.
But mounting this xfs filesystem in a mount point failed permanently. no matter what i tried.
Then I found a bug fix on the internet that apparently fixes this. However, this solution requires a new compilation of the kernel.

Because I don't want to wait until this problem is resolved, I installed the 32-bit version and the xfs-formatted xfs partition can be mounted without any problems.

I can no longer deliver error messages because I have now installed the 32Bit system. But try it yourself.

PS: why don't you compile the 64 bit kernel with the same basic configuration as the 32 kernel ???

Translated with Google

XECDesign, 2022-02-16:
@popcornmix @pelwell sounds like a kernel config issue?

popcornmix, 2022-02-16:
It's quite an old issue from a deleted github account, so I don't think we'll get any more information.

pelwell, 2022-02-16:
But for the record, the ARCH=arm kernel has:
```
# Linux/arm 5.15.23 Kernel Configuration
...
CONFIG_XFS_FS=m
CONFIG_XFS_SUPPORT_V4=y
CONFIG_XFS_QUOTA=y
CONFIG_XFS_POSIX_ACL=y
CONFIG_XFS_RT=y
# CONFIG_XFS_ONLINE_SCRUB is not set
# CONFIG_XFS_WARN is not set
# CONFIG_XFS_DEBUG is not set
# CONFIG_VXFS_FS is not set
```
while the arm64 kernel has:
```
# Linux/arm64 5.15.23 Kernel Configuration
...
CONFIG_XFS_FS=m
CONFIG_XFS_SUPPORT_V4=y
CONFIG_XFS_QUOTA=y
CONFIG_XFS_POSIX_ACL=y
CONFIG_XFS_RT=y
# CONFIG_XFS_ONLINE_SCRUB is not set
# CONFIG_XFS_WARN is not set
# CONFIG_XFS_DEBUG is not set
# CONFIG_VXFS_FS is not set
```
I don't see a lot of difference there.

pelwell, 2022-02-16:
I don't imagine 5.10 was any different.

popcornmix, 2022-02-16:
Closing. Can be reopened if there's any new information.

closed 2022-02-16 by popcornmix