raspberrypi/Raspberry-Pi-OS-64bit #181: cannot open Sonic Pi on  Raspberry Pi OS

edmundlaugasson, 2021-09-26:
I refer to sonic-pi-net/sonic-pi#2803 which was initially posted to Sonic Pi repository, but they do not prepare Raspberry Pi OS. Hopefully can you look into it and finally fix it.

ghollingworth, 2021-09-26:
You should ask the maintainers of the software application to look into this.

closed 2021-09-26 by ghollingworth

edmundlaugasson, 2021-09-26:
Who is maintainer? I thought you are packaging software in Raspberry Pi OS. Did you read the referred issue? This is software maintainer's repository already. They say, that they are not responsible of packaging software in Raspberry Pi OS. Therefore I came here. Now you say, that you don't package software either. It's easy to say, I don't deal with it, but in such case - who does?

ghollingworth, 2021-09-26:
Ah, Ok we'll remove it from our repos

edmundlaugasson, 2021-09-26:
I downloaded from Sonic Pi GitHub repository - then it works. Downloading Rasperry Pi OS and trying to run Sonic Pi - it doesn't. I understood, that you as OS maintainer do packaging. Perhaps you should contact with Sonic Pi developers to get this properly packaged. I'm just regular user, who noticed, that bundled Sonic Pi in Rasperry Pi OS doesn't work and decided to say it also here.

rbnpi, 2021-09-30:
With regards to the above, I am a member of the Sonic Pi development team, and I have been cooperating with Simon Long to adjust the Sonic Pi 3.2.2 source deb file currently in Debian Bullseye to produce a fully working version for the next Raspberry Pi OS when that is released. The Debian version is produced by and maintained by the Debian multimedia team, and presumably the patched version Simon has produced will be the responsibility of the Raspberry Pi OS maintainers. Sonic Pi is now in beta for version 4.0, and is currently built using a cmake script, but this is not setup to produce a standard source deb file. We do not have the resources to undertake this task, but will continue to produce binary deb files for those who want to use the latest versions. Currently we have internal builds for this which work on a range of linux distros, as well as windows PC and Mac.
We are aware of the current situation and are sorry for the confusion and agro the lack of up to date debs has caused, and for the fact that we do not have the resources to overcome this, although we would welcome any assistance in this area - especially from those with the expertise of producing source debs for Debian.