raspberrypi/Raspberry-Pi-OS-64bit #183: libsdl2-dev not installing

Botspot, 2021-09-30:
This is a fresh installation of Raspberry Pi OS Buster 64-bit.
```
pi@raspberrypi:~$ sudo apt install libsdl2-dev -yf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libsdl2-dev : Depends: libsdl2-2.0-0 (= 2.0.9+dfsg1-1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
Unless I'm mistaken, this error did not used to occur.

XECDesign, 2021-09-30:
Looks okay on this end:

```
# apt install libsdl2-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  gir1.2-ibus-1.0 libasound2-dev libblkid-dev libdbus-1-dev libegl-dev libegl1-mesa-dev libffi-dev libgl-dev libgl1-mesa-dev libgles-dev libgles2-mesa-dev libglib2.0-dev libglib2.0-dev-bin libglu1-mesa libglu1-mesa-dev libglx-dev libibus-1.0-5
  libibus-1.0-dev libice-dev libmount-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpthread-stubs0-dev libpulse-dev libpulse-mainloop-glib0 libselinux1-dev libsepol1-dev libsm-dev libsndio-dev libudev-dev libwayland-bin libwayland-dev
  libx11-dev libxau-dev libxcb1-dev libxcursor-dev libxdmcp-dev libxext-dev libxfixes-dev libxi-dev libxinerama-dev libxkbcommon-dev libxrandr-dev libxrender-dev libxss-dev libxt-dev libxv-dev libxxf86vm-dev uuid-dev x11proto-core-dev x11proto-dev
  x11proto-fixes-dev x11proto-input-dev x11proto-randr-dev x11proto-scrnsaver-dev x11proto-xext-dev x11proto-xf86vidmode-dev x11proto-xinerama-dev xorg-sgml-doctools xtrans-dev
Suggested packages:
  libasound2-doc libglib2.0-doc libgdk-pixbuf2.0-bin | libgdk-pixbuf2.0-dev libxml2-utils libice-doc libsm-doc libwayland-doc libx11-doc libxcb-doc libxext-doc libxt-doc
The following NEW packages will be installed:
  gir1.2-ibus-1.0 libasound2-dev libblkid-dev libdbus-1-dev libegl-dev libegl1-mesa-dev libffi-dev libgl-dev libgl1-mesa-dev libgles-dev libgles2-mesa-dev libglib2.0-dev libglib2.0-dev-bin libglu1-mesa libglu1-mesa-dev libglx-dev libibus-1.0-5
  libibus-1.0-dev libice-dev libmount-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpthread-stubs0-dev libpulse-dev libpulse-mainloop-glib0 libsdl2-dev libselinux1-dev libsepol1-dev libsm-dev libsndio-dev libudev-dev libwayland-bin
  libwayland-dev libx11-dev libxau-dev libxcb1-dev libxcursor-dev libxdmcp-dev libxext-dev libxfixes-dev libxi-dev libxinerama-dev libxkbcommon-dev libxrandr-dev libxrender-dev libxss-dev libxt-dev libxv-dev libxxf86vm-dev uuid-dev x11proto-core-dev
  x11proto-dev x11proto-fixes-dev x11proto-input-dev x11proto-randr-dev x11proto-scrnsaver-dev x11proto-xext-dev x11proto-xf86vidmode-dev x11proto-xinerama-dev xorg-sgml-doctools xtrans-dev
0 upgraded, 63 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,936 kB of archives.
After this operation, 43.9 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

You could try `aptitude` instead of `apt`, which is a bit smarter and tries harder to resolve issues. Or try installing libsdl2-2.0-0 and see what it conflicts with.

pelwell, 2021-09-30:
There's a mixture of explanation and empathy here: https://itsfoss.com/held-broken-packages-error/