raspberrypi/Raspberry-Pi-OS-64bit #184: Problem with installing xrdp

Knallkopf97, 2021-10-02:
Hey :)

I have a problem with installing xrdp on Rasp 64bit OS (beta). 
i use the comand:
- sudo apt-get install xrdp

This is what i get:

Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-colorzero rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libglu1-mesa ssl-cert x11-apps x11-session-utils xbitmaps xfonts-100dpi xfonts-75dpi xfonts-base xfonts-encodings xfonts-scalable xfonts-utils xorg xorg-docs-core xorgxrdp
Suggested packages:
  openssl-blacklist mesa-utils xorg-docs x11-xfs-utils guacamole xrdp-pulseaudio-installer
The following NEW packages will be installed:
  libglu1-mesa ssl-cert x11-apps x11-session-utils xbitmaps xfonts-100dpi xfonts-75dpi xfonts-base xfonts-encodings xfonts-scalable xfonts-utils xorg xorg-docs-core xorgxrdp xrdp
0 upgraded, 15 newly installed, 0 to remove and 2 not upgraded.
Need to get 417 kB/15.6 MB of archives.
After this operation, 24.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://deb.debian.org/debian buster/main arm64 xrdp arm64 0.9.9-1
  404  Not Found [IP: 199.232.190.132 80]
E: Failed to fetch http://deb.debian.org/debian/pool/main/x/xrdp/xrdp_0.9.9-1_arm64.deb  404  Not Found [IP: 199.232.190.132 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
______________________________________________________________________________________________________________________

I have the same problem after upgrade RaspOS with following command:
- sudo apt-get update
- sudo apt get upgrade
- sudo reboot

Any Ideas how to fix it?

lurch, 2021-10-02:
Looking at http://ftp.debian.org/debian/pool/main/x/xrdp/ the nearest filename-match seems to be http://ftp.debian.org/debian/pool/main/x/xrdp/xrdp_0.9.9-1+deb10u1_arm64.deb so perhaps something's gone "wonky" with one of the Debian package lists? :man_shrugging:

XECDesign, 2021-10-03:
This is what you're after: https://deb.debian.org/debian/pool/main/x/xrdp/xrdp_0.9.9-1+deb10u1_arm64.deb

What's the output of `apt policy`?