raspberrypi/Raspberry-Pi-OS-64bit #189: Stuck at rainbow screen on Pi Zero 2 W

lberki, 2021-11-03:
After preparing an SD card with the image `2021-05-07-raspios-buster-arm64-lite.img` (SHA256 of ZIP verified as `868cca691a75e4280c878eb6944d95e9789fa5f4bbce2c84060d4c39d057a042`), The Zero 2 W get stuck at the rainbow boot screen.

The same device with the same SD card works correctly when flashed with the `armhf` image. I did not verify that the SD card was written correctly, but I tried this twice and it would be strange if it flaked out twice in the same way.

The command I used to prepare the SD card is the `sudo dd if=2021-05-07-raspios-buster-arm64-lite.img of=/dev/sda bs=4M status=progress conv=fsync && sudo sync`

lberki, 2021-11-03:
Update: `cp bcm2710-rpi-3-b.dtb bcm2710-rpi-zero-2.dtb` on the boot partition makes it work.

I'll leave this bug open since it should probably work on an intact 64-bit image, too.

lurch, 2021-11-03:
If you boot that card in a non-Zero2W system, do a `sudo apt update && sudo apt upgrade`, shut down and swap the card back to the Zero2W, it _should_ then boot successfully?

pelwell, 2021-11-03:
The behaviour is expected. At some point we'll update the 64-bit images, but that's more likely to be once we've launched Bullseye. I'm the meantime, do as lurch said, or download and install bcm2710-rpi-zero-2.dtb from here: https://github.com/raspberrypi/firmware/blob/master/boot/bcm2710-rpi-zero-2.dtb

lberki, 2021-11-03:
I don't have another easily accessible Raspberry Pi to do the `apt update` / `apt upgrade` trick. Instead, what I did:

1. Boot with the above `cp bcm2710-rpi-3-b.dtb bcm2710-rpi-zero-2.dtb` trick
2. `mv bcm2710-rpi-zero-2.dtb bcm2710-rpi-zero-2.dtb.orig`
3. `apt update && apt upgrade`

At this point, there was no `bcm2710-rpi-zero-2.dtb` under /boot and as expected, the boot process got stuck at the rainbow screen stage.

Then I tried the file @pelwell recommended above (SHA 256: `7a34c01451394948e1b72ff9d0401b93ec52a696d31c528298392622e293b2d7`), and this time, it booted, but at some point in the boot process (I don't remember where), the HDMI signal was lost. I didn't check whether the device was pingable (tried this twice, so I don't think it's a random occurrence)

Going back to the .dtb file of the 3B (as in step 1 above) fixed the issue. `/var/log/syslog` contains a bunch of 0 bytes around the interesting times, so I'm afraid that's not of much help.

I'm not sure if that's the right package, but `raspberrypi-bootloader` has version `1:1.20211007-2~buster`, if that helps.

lurch, 2021-11-04:
> I'm not sure if that's the right package, but raspberrypi-kernel has version 1:1.20211007-2~buster, if that helps.

That date is from before the public announcement of the RPi Zero 2 W, so I guess it's not surprising that `bcm2710-rpi-zero-2.dtb` isn't present :wink:  @XECDesign Will `bcm2710-rpi-zero-2.dtb` be appearing in an apt version of raspberrypi-kernel (or raspberrypi-firmware?) soon?

XECDesign, 2021-11-04:
Not before it appears here: https://github.com/raspberrypi/firmware/tree/stable/boot and not before the initial Bullseye release, so I wouldn't say soon.

lurch, 2021-11-04:
Ahh, I didn't realise that the apt packages were now built from a `stable` branch of the firmware repo - don't think I even noticed that branch before :laughing: 

Still, it seems odd that an `apt`-upgraded 64-bit RasPiOS isn't booting on a Zero 2 W, when presumably it works fine in the 32-bit OS?

JamesH65, 2021-11-04:
I believe it needs specific DTB for 64bit Pi02  that is not yet in the distro?

XECDesign, 2021-11-04:
It needs either the DTB Phil linked to or the old zero DTB. The 64-bit release ships neither.

l00nix, 2021-11-05:
Thanks for all the advice to get 64-bit raspios booting on the pi zero 2. I am still having issues getting it up and running fully. I followed the following steps:

1. imaged [2021-05-07-raspios-buster-arm64-lite.zip](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip) to SD card using Raspberry Pi Imager.
2. booted in SD card on a RPi model 4B
3. updated with `apt-get update && apt-get upgrade`
4. copied [bcm2710-rpi-zero-2.dtb](https://github.com/raspberrypi/firmware/blob/master/boot/bcm2710-rpi-zero-2.dtb) to /boot
5. booted SD in pi zero 2

I can now get past the rainbow screen and it starts booting but after the kernel starts loading I get a blank screen after the following:

```
...
[ OK ] Started udev Kernel Device Manager
[ OK ] Founf device /dev/disk/by-partuuid/d953c604-01.
```

Any ideas what I am missing?

lberki, 2021-11-05:
Sounds like my experience with the aforementioned `.dtb` file. What I ended up doing was `cp bcm2710-rpi-3-b.dtb bcm2710-rpi-zero-2.dtb` on an intact Raspbian 64-bit image. That worked (or at least I could log in and make WiFi interface work)

l00nix, 2021-11-05:
@lberki - you are the man - thanks, 

> Sounds like my experience with the aforementioned `.dtb` file. What I ended up doing was `cp bcm2710-rpi-3-b.dtb bcm2710-rpi-zero-2.dtb` on an intact Raspbian 64-bit image. That worked (or at least I could log in and make WiFi interface work)

That worked for me

Alex

pelwell, 2021-11-08:
A recent change to allow the VEC clock to be controlled from Linux has made the current DTB incompatible with older kernels and firmwares. For me the kernel boots but the display remains blank, with the following messages appearing in the kernel log:
```
[   10.190698] of_clk_hw_onecell_get: invalid index 15
[   10.190787] [drm:vc4_vec_bind [vc4]] *ERROR* Failed to get clock: -2
[   10.190893] vc4-drm soc:gpu: failed to bind 3f806000.vec (ops vc4_vec_ops [vc4]): -2
```
A custom Zero 2 W dtb with that DT change reverted can be downloaded from here: https://drive.google.com/file/d/11B6fPFdE3W5lKwrwmYYWgbv6KmDDg0Nl/view?usp=sharing

mickeyl, 2021-11-09:
A [new snapshot](http://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-11-08/) has appeared -- unfortunately copying the device tree file seems to be still necessary.

Coolykoen, 2021-11-15:
i am completely new to this, how do i copy the dtb? i get the rainbow screen aswell.
Should i get the dtb from another pi image? and where to i put it on my sd card for the zero 2 w?

mickeyl, 2021-11-15:
@Coolykoen After flashing the SD image, you need to mount it on a computer that can read Linux filesystems. You will then see two filesystems, one `/boot`, and one `rootfs`. On the former, copy as [instructed here](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/189#issuecomment-962064297). Unmount the SD and then you should be able to boot your Raspberry Pi Zero 2 for the very first time.

Coolykoen, 2021-11-15:
oh... the pi is my first linux running hardware hmm
i do have past experience with desktop linux distros but my main computer is just a Windows install.

i'll see if i can manage to forward the sd reader to a vm running linux haha

Thanks for the info :)

lurch, 2021-11-15:
You don't _need_ a Linux computer - the `/boot` partition is FAT32-formatted, so can be read by any computer :slightly_smiling_face: https://www.raspberrypi.com/documentation/computers/configuration.html#the-boot-folder

Coolykoen, 2021-11-17:
I was able to boot correctly by putting the file from this comment in the boot folder:
https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/189#issuecomment-963027691

Thanks everyone :3

springjools, 2021-11-23:
What .dtb image do I put on a Pi zero-w that has the same symptom? Can I use a 0-2 dtb?

pelwell, 2021-11-23:
No - there are many things that can stop a Zero W from booting, but asking it to use a Zero 2 W dtb is another one. Your problem is caused by something else altogether.

lurch, 2021-11-23:
The Pi Zero W only has a 32-bit CPU, so will be unable to run the 64-bit Raspberry Pi OS (which is what this repo is all about).

matthew5025, 2021-11-23:
This issue refers to the Pi Zero 2 which has a 64 bit cpu.

vFloresHx, 2021-11-26:
I'm still stuck, my zero2 won't get past the rainbow, this is after the apt update && upgrade, and having tried with several different dtb files, including copy/pasting the pi3 one and renaming it, the one github and the one on this thread. 

The image i'm trying to run is a copy of the one I'm currently running on my pi4 into another SD card, which boots OK in the pi4 but won't boot on the zero2. The pi however, boots with the included Noobs SD card, so any hardware issue is discarded, it's just something with my image. 

Does anyone have any idea whatsoever of what could be causing the zero2 not to boot an image that's running ok on a pi4?

VirusPilot, 2021-12-02:
Here is what steps worked in my case for a Pi Zero 2 W:

1. image [http://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-11-08/2021-10-30-raspios-bullseye-arm64-lite.zip](url) on SD card
2. copy empty `ssh` file to `/boot`
3. boot SD card in a Pi4B
4. update with `apt update && apt full-upgrade`
5. remove SD card
6. copy `wpa_supplicant.conf` file to `/boot` (with your WiFi credentials)
7.  boot SD card in a Pi Zero 2 W

lurch, 2021-12-03:
https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/ says "the 64bit OS, that’s coming next, we’re just fixing some issues in Bullseye before we release it", so no doubt that'll work out-of-the-box with the Pi Zero 2 W (with no mucking about with DTB files needed).

mickeyl, 2021-12-03:
Hopefully this includes kernel 5.15.

pelwell, 2021-12-03:
You're a bit previous - the aim is to get stable Bullseye images on 5.10 out now-ish, the 5.15 early next year.