raspberrypi/Raspberry-Pi-OS-64bit #191: DHCP ipv4 not being taken on eth0

robertalexa, 2021-11-04:
Hi,

I have been experiencing this really annoying situation that I have not been able to get to the bottom of. In short, on boot (sporadically but very very often), the rPi is not getting the ipv4 assigned on the eth0 interface.

Here are my findings and thoughts:
* There is no dhcp/networking changes on the rPi, everything is default as found on the latest beta image, so we know there is no "user interference".
* The DHCP server (in my case a Unifi UDM Pro) is giving a fixed ip to this rPi (so we know there is no dhcp ip conflict).
* There are no obvious errors
* Upon boot, if i run `sudo dhclient eth0` i get the correct IP instantly and everything works as expected.
* As mentioned, this issue manifests on most reboots, but very rarely it doesn't e.g. i reboot and everything works
* Another thing to mention is that I am running docker containers on this machine, i did read somewhere that this could "confuse" dhcp but I can't seem to find where i read that.

As an interim solution I am basically running a cronjob as root that does `@reboot sleep 10 \usr\sbin\dhclient eth0` and that gets me up an running.

Any advice would be greatly appreciated. I would be more than happy to provide more details/logs if you can tell me what would be relevant in this context.

Thanks

lurch, 2021-11-04:
> Another thing to mention is that I am running docker containers on this machine, i did read somewhere that this could "confuse" dhcp but I can't seem to find where i read that.

#57 ?

robertalexa, 2021-11-04:
you kind sir are an angel in disguise. That feels like it has done the trick.

Nonetheless, if you indulge me, any thoughts on why that is an issue?

In my case, I have another rpi 4 running raspbian x32 running more than twice docker containers than this one (x64) and that has never behaved like this. On top of that, I have just set up 4 x rPi running ubuntu x64 which run docker containers as their sole purpose, and they are not having this issue either?

I would be curios to understand more, whether this is an actual true issue with x64 or if it is a "limitation" of dhcp client?

Nonetheless, thank you once more for your help.

PS: i will be testing this in more depth in the following days, but so far everything works. If anyone else stumbles upon this. Edit `/etc/dhcpcd.conf` and add `denyinterfaces veth*` at the bottom.

lurch, 2021-11-04:
I think RPiOS 64-bit might be more closely-aligned to upstream Debian (i.e. has fewer RPi-specific changes) than RPiOS 32-bit is? :shrug:

robertalexa, 2021-11-04:
Thanks for your input, however I am not sure I understand the corelation? Are you suggestion that this dhcp situation is a "real problem" when using docker containers on any OS? Like some sort of racing condition for dhcp and to who does dhcp assign the ip or sth?

lurch, 2021-11-04:
I've got no idea - I simply found that other issue by searching for docker in this repo: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/search?q=docker&type=issues

robertalexa, 2021-11-04:
Understood, well thanks for pointing me in the right direction :)

closed 2021-11-04 by robertalexa