raspberrypi/Raspberry-Pi-OS-64bit #192: wrong condition in alsa-state.service

quantenschaum, 2021-11-06:
In `/lib/systemd/system/alsa-state.service` the condition line is

```
ConditionPathExists=!/etc/alsa/state-daemon.conf
```
 but it should be negated.

```
ConditionPathExists=/etc/alsa/state-daemon.conf
```

pelwell, 2021-11-06:
How does the system malfunction without this change?

XECDesign, 2021-11-06:
Doesn't looks like a 64-bit issue or a Raspberry Pi OS issue.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=925455
https://salsa.debian.org/alsa-team/alsa-utils/commit/af161676131e94bbaed72f37d0c5d4c6685a119e

Then there was another commit that didn't make it into the buster release that does things a bit differently:
https://salsa.debian.org/alsa-team/alsa-utils/-/commit/da4803a0c88e83a0e580c98f1474872d1f6288bf

Without fully understanding the issue, I wouldn't go ahead overriding what Debian has settled on.

quantenschaum, 2021-11-09:
Yes, probably the wrong issue tracker, but I just found this one. Where is the RaspiOS 32 repo?

@pelwell Anyways, with the setting as it is in the distro, `alsa-state.service` runs, but it should not, because `alsa-restore.service` is doing the job. The presense of `/etc/alsa/state-daemon.conf` should disable `alsa-restore.service` and enable `alsa-state.service` . Now *both* or *none* is running.