raspberrypi/Raspberry-Pi-OS-64bit #196: Original Touch Screen is not working with 64bit on Pi 3B

ralf-9000, 2021-12-01:
Hi,
I am trying to update my Touchscreen-Pi's with the 64bit image, the original 7" touchscreen is not working, only light rainbow pattern is visible. A new setup with the 32bit image is working fine. Have I to do a additional setup?
Ralf

lurch, 2021-12-01:
Does this help? https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/185#issuecomment-952219630

6by9, 2021-12-01:
I've lost track of what has been released for the 64bit OS.
If 64bit has been updated in the same way as Bullseye 32 bit, then I'd expect to see ```display_auto_detect=1``` in config.txt. If so then adding ```dtoverlay=vc4-kms-dsi-7inch``` is not necessary.

More likely https://github.com/raspberrypi/linux/issues/4686. Add ```dtparam=i2c_vc_baudrate=50000``` to config.txt.

ralf-9000, 2021-12-01:
Many thanks, 6by9 !!!

`dtparam=i2c_vc_baudrate=50000` has helped, works now!

`display_auto_detect=1` was already set and adding `dtoverlay=vc4-kms-dsi-7inch` has not helped. Adding the baudrate switch only was the key.

Ralf

closed 2021-12-01 by ralf-9000