raspberrypi/Raspberry-Pi-OS-64bit #198: Alt+Tab toggles between only two Applications

RaviRJoshi, 2021-12-14:
If you have more than two windows open, alt+tab toggles only between current & next most recent window.
This issue is not specific to Oct 2021 beta release. It happens in earlier versions, too.

pelwell, 2021-12-14:
This should probably be filed in https://github.com/raspberrypi-ui/lxpanel-bullseye (it's certainly not specific to the 64-bit OS) - @spl237?

spl237, 2021-12-14:
Looks like a bug in mutter - shift-alt-tab works fine, and the shortcuts are correctly assigned to switch-applications and switch-applications-backward. Not something I am aware I have changed, so filing a report against the upstream mutter repo is probably the best option - https://github.com/GNOME/mutter

ZanderBrown, 2021-12-14:
That is _not_ the upstream repo → [this is](https://gitlab.gnome.org/GNOME/mutter/)

Additionally it's worth noting `switch-applications` switches _applications_ (not windows), and you can within an application (`switch-group`) with <kbd>super</kbd>-<kbd>`</kbd>.

For a more windows-esc experience there is `switch-windows`, but it's unbound by default

Drdrm46, 2022-01-22:
I don't think this a bug in mutter because Alt Tab works perfectly in GNOME which uses gnome-shell to augment mutter functionality including Alt Tab and associated application selection icons. File against github Rpi-distro raspberrypi-ui-mods

Drdrm46, 2022-01-25:
This has now been filed against Rpi-distro raspberrypi-ui-mods.