raspberrypi/Raspberry-Pi-OS-64bit #199: cubicSDR and the new Raspberry Bullseye (bugs)

raspi-rick, 2021-12-14:
Hello. Today I installed the new Raspberry OS (Bullseye) in my Raspberry Pi 4B. Then I installed CubicSDR v0.2.5 from the repository. This application has a bug that did not happen in the previous version of Raspberry OS. For example, to set the frequency of 150.1 MHz, the input jumps from 149.1 to 151.1 This problem did not happen with the previous version of Raspberry OS, even if the version of CubicSDR is the same. I hope you can fix this bug in the next update of Raspberry OS. Thank you. | Hello. Today I installed the new Raspberry OS (Bullseye) in my Raspberry Pi 4B. Then I installed CubicSDR v0.2.5 from the repository. This application has a bug that did not happen in the previous version of Raspberry OS. For example, to set the frequency of 150.1 MHz, the input jumps from 149.1 to 151.1 This problem did not happen with the previous version of Raspberry OS, even if the version of CubicSDR is the same. I hope you can fix this bug in the next update of Raspberry OS. Thank you. PS: another bug is that "start recording" and "stop recording" appear in light gray background with white fonts, which is hard to read. 
-- | --
Hello. Today I installed the new Raspberry OS (Bullseye) in my Raspberry Pi 4B. Then I installed CubicSDR v0.2.5 from the repository. This application has a bug that did not happen in the previous version of Raspberry OS. For example, to set the frequency of 150.1 MHz, the input jumps from 149.1 to 151.1 This problem did not happen with the previous version of Raspberry OS, even if the version of CubicSDR is the same. I hope you can fix this bug in the next update of Raspberry OS. Thank you. PS: another bug is that "start recording" and "stop recording" appear in light gray background with white fonts, which is hard to read.

<img height="1" width="1" class="yiv9180637054ycb2701357601" style="font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none; border: 0px; text-indent: -9999px; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Roboto, &quot;Helvetica Neue&quot;, Arial, sans-serif; font-size: 14px;">

JamesH65, 2021-12-14:
I've edit the OP - it had some really weird formatting in it. Content remains the same.

lurch, 2021-12-15:
I suspect this issue probably belongs at https://github.com/RPi-Distro/repo/issues rather than in the 64-bit-beta-issues repo?

raspi-rick, 2021-12-15:
this was posted at the request of Kitti at raspberry.com. See below. 

Hi Rick,
Thank you for getting in touch. 
​Please could you report the issue on the github forums at https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues
​Kind regards, 
​Kitti

JamesH65, 2021-12-15:
Hmm, tried to transfer the issues, but moving to RPi-Distro does not seem possible, perhaps different organisations?

JamesH65, 2021-12-15:
With regard to the issue, I think this is actually an issue with CubicSDR anyway, we don't have any in house knowledge of it, so are very unlikely to look at it. I'd suggest reporting to the CubicSDR devs.

XECDesign, 2021-12-15:
Yeah, this doesn't sound like a Raspberry Pi OS issue. We don't modify that package in any way, so the issue would need to go upstream or to Debian (assuming it also occurs there). If there's a patch that fixes the issues, we could pull it in.