raspberrypi/Raspberry-Pi-OS-64bit #200: raspi-config resolution menu won't show

BenMcLean, 2021-12-28:
I'm actually using RetroPie on my Raspberry PI 400 but I believe this issue is actually with Raspberry PI OS because it is `raspi-config` which is acting up.

In `raspi-config`, when I go to "Resolution" in the "Display" menu in raspi-config then it just immediately quits and goes back to the "Display" menu.

I think this may have to do with the fact that the resolution is 4K. I was going in to that menu to try to reduce it to 1080p. Apparently, the way to do that on RetroPie isn't available from a menu right now: you have to edit the `config.txt` for that.

However, if raspi-config's resolution menu needs to stop for some reason then it really should display a proper error message to the user, right?

Seemed like I should send in an issue for this at least.

BenMcLean, 2021-12-29:
Looks like I have made a duplicate of [this](https://github.com/RPi-Distro/raspi-config/issues/155).

spl237, 2022-02-28:
This looks to be a RetroPie issue. Recent builds of Raspberry Pi OS use the KMS video driver, which is not compatible with the resolution setting in raspi-config; the command-line tool xrandr should be used to set resolution instead. On my latest image, this menu option is not even shown; it should be hidden with the KMS driver in use. I don't know why the menu option is being shown on RetroPie - it may be that the authors have included an older version of the tool for some reason. But it won't work - xrandr is now the way to do this.

BenMcLean, 2022-02-28:
> This looks to be a RetroPie issue. Recent builds of Raspberry Pi OS use the KMS video driver, which is not compatible with the resolution setting in raspi-config; the command-line tool xrandr should [...]

So ... couldn't `raspi-config` be updated to act as a front-end for `xrandr`?

spl237, 2022-02-28:
Yes, it could, but only if I (or someone else) had time to do it. It's much easier for people to just use xrandr directly.

BenMcLean, 2022-03-09:
> Yes, it could, but only if I (or someone else) had time to do it. It's much easier for people to just use xrandr directly.

If we're not going to fix the Resolution menu and we know the Resolution menu should never work, then the fix should be to remove the Resolution menu from `raspi-config` altogether, right?

spl237, 2022-03-09:
> If we're not going to fix the Resolution menu and we know the Resolution menu should never work, then the fix should be to remove the Resolution menu from `raspi-config` altogether, right?

Already done for the next release.

BenMcLean, 2022-03-09:
> Already done for the next release.

OK thanks. :)

closed 2022-03-09 by BenMcLean