raspberrypi/Raspberry-Pi-OS-64bit #201: Veracrypt device-mapper issue

Marco-Total, 2022-01-21:
Hi,

after update my Rasperry Pi OS to Bullseye my veracrypt installation struggle.

```
root@mapi:/home/pi# cat /etc/issue
Debian GNU/Linux 11 \n \l
```
```

root@mapi:/home/pi# uname -r
5.10.63-v8+
```

```
root@mapi:/home/pi# veracrypt --version
VeraCrypt 1.25.4
```

```
root@mapi:/home/pi# veracrypt --mount --fs-options=umask=000 --non-interactive --protect-hidden=no --password='XXX' /dev/disk/by-partuuid/fefadefd-4e55-402a-9c03-50632421c79e /media/Daten
Error: /dev/mapper/control: open failed: Kein passendes Gerät gefunden
Failure to communicate with kernel device-mapper driver.
Check that device-mapper is available in the kernel.
Incompatible libdevmapper 1.02.175 (2021-01-08) and kernel driver (unknown version).
Command failed.
```

Cheers, Marco

pelwell, 2022-01-21:
Which is the most recent kernel version for which Veracrypt works?

pelwell, 2022-01-21:
And what do the following commands return?
```
$ uname -a
$ sudo modprobe configs
$ zgrep BLK_DEV_DM /proc/config.gz
```

pelwell, 2022-01-21:
And:
```
$ ls /dev/disk/by-partuuid/
```

Marco-Total, 2022-01-21:
Hi,

i can only see, that veracrypt run without this error on 2021-11-05, i run every week a backup script.

```
root@mapi:/home/pi# uname -a
Linux mapi 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux
```

```
root@mapi:/home/pi#  sudo modprobe configs
modprobe: FATAL: Module configs not found in directory /lib/modules/5.10.63-v8+
```

```
root@mapi:/home/pi# zgrep BLK_DEV_DM /proc/config.gz
gzip: /proc/config.gz: No such file or directory
```

```
root@mapi:/home/pi# ls /dev/disk/by-partuuid/
091c4cb6-3e95-4ac4-882e-3c3448b2b9d9  15f86db9-01  34b5e3ee-02                           649a3bd7-9f7a-47b6-9c59-21bc185f945d  d902bbc2-1295-ab4c-896d-9ec7a9d2b095
15c45998-e0d5-4d7a-8fa0-46e7cdaa3ab2  34b5e3ee-01  4f4536da-3e1b-42b3-a0bb-e763d173911c  98d31ed7-1e57-497c-911d-ea785c81292f  **fefadefd-4e55-402a-9c03-50632421c79e**
```

pelwell, 2022-01-21:
Something has gone wrong with your installation - this is what I get running the same kernel build:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.10.63-v8+ #1496 SMP PREEMPT Wed Dec 1 15:59:46 GMT 2021 aarch64 GNU/Linux
pi@raspberrypi:~$ sudo modprobe configs
pi@raspberrypi:~$ zgrep BLK_DEV_DM /proc/config.gz
CONFIG_BLK_DEV_DM_BUILTIN=y
CONFIG_BLK_DEV_DM=m
```
Try running `sudo rpi-update 05dc06d` to re-install the same kernel and firmware, but first check you haven't run out of disk space.

popcornmix, 2022-01-21:
Do you reboot after updating? If kernel updates that is needed.

Marco-Total, 2022-01-21:
Thank you! rpi-update 05dc06d solve my problems! 

I think i should update my kernel with apt upgrade? In wich cases i should execute rpi-updade?

```
root@mapi:/home/pi# uname -r
5.10.63-v8+
root@mapi;/home/pi# sudo modprobe configs
root@mapi:/home/pi# zgrep BLK_DEV_DM /proc/config.gz
CONFIG_BLK_DEV_DM_BUILTIN=y
CONFIG_BLK_DEV_DM=m
```

pelwell, 2022-01-21:
Does that mean Veracrypt is working now? If so, the problem was just that the kernel couldn't find a compatible device mapper module.

`rpi-update` was an attempt to get you back to working system while changing as few things as possible. Under normal circumstances you should only `apt update/upgrade/full-upgrade`. You could try the following command to put you back on the normal apt install track:
```
$ sudo apt upgrade --reinstall raspberrypi-kernel raspberrypi-bootloader
```
Alternatively or you can just wait until the next kernel and bootloader releases, and `apt upgrade` as normal.