raspberrypi/Raspberry-Pi-OS-64bit #202: raspi 3b can't update kernel

zewenhs, 2022-01-26:
I download this 64 bit os:https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/, it can boot normally on raspi 3b:
```shell
Linux raspberrypi 5.4.42-v8+ #1319 SMP PREEMPT Wed May 20 14:18:56 BST 2020 aarch64 GNU/Linux
```

Because I need to do some work with kernel, so I follow [Cross-Compiling the Raspi Kernel](https://www.raspberrypi.com/documentation/computers/linux_kernel.html) to build the kernel(v5.4) and install to SD card, **Nothing** will be display after the rainbow screen, so my question is how to update kernel on raspi 3b 64 bit kernel??

pelwell, 2022-01-26:
Just follow that guide, being careful to use the 64-bit sections. Although the guide says to use bcmrpi3_defconfig on a 3/3+, the 64-bit kernel we ship is built from the bcm2711_defconfig and runs on all ARMv8 RPis.

lurch, 2022-01-26:
> Although the guide says to use bcmrpi3_defconfig on a 3/3+, the 64-bit kernel we ship is built from the bcm2711_defconfig and runs on all ARMv8 RPis.

Would you mind submitting an appropriate [PR](https://github.com/raspberrypi/documentation/pulls) please Phil?

pelwell, 2022-01-26:
https://github.com/raspberrypi/documentation/pull/2381

zewenhs, 2022-01-27:
It works, good!

closed 2022-02-07 by zewenhs