raspberrypi/Raspberry-Pi-OS-64bit #203: "OpenMAX IL headers from raspberrypi/firmware not found" error on ffmpeg compilation

dbuscaino, 2022-01-26:
Hi everyone, I'm trying to compile ffmpeg on Raspberry Pi3 following this guide [compiling-ffmpeg-raspberry-pi](https://pimylifeup.com/compiling-ffmpeg-raspberry-pi/).
After the compilation of the additional libraries when I try to run ./configure for ffmpeg I get "OpenMAX IL headers from raspberrypi/firmware not found" error cause by --enable-omx-rpi option.

Searching on internet ([link](https://stackoverflow.com/questions/40175644/ffmpeg-hardware-acceleration-on-raspberry-pi)) I found that --enable-omx-rpi requires libomxil-bellagio-dev (Ubuntu) package which provides the required header file OMX_Core.h.

I installed libomxil-bellagio-dev with sudo apt-get install libomxil-bellagio-dev. Running dpkg -L libomxil-bellagio-dev it give me:

`/.
/usr
/usr/include
/usr/include/OMX_Audio.h
/usr/include/OMX_Component.h
/usr/include/OMX_ContentPipe.h
/usr/include/OMX_Core.h
/usr/include/OMX_IVCommon.h
/usr/include/OMX_Image.h
/usr/include/OMX_Index.h
/usr/include/OMX_Other.h
/usr/include/OMX_Types.h
/usr/include/OMX_Video.h
/usr/include/bellagio
/usr/include/bellagio/OMXComponentRMExt.h
/usr/include/bellagio/OMXCoreRMExt.h
/usr/include/bellagio/common.h
/usr/include/bellagio/component_loader.h
/usr/include/bellagio/extension_struct.h
/usr/include/bellagio/omx_base_audio_port.h
/usr/include/bellagio/omx_base_clock_port.h
/usr/include/bellagio/omx_base_component.h
/usr/include/bellagio/omx_base_filter.h
/usr/include/bellagio/omx_base_image_port.h
/usr/include/bellagio/omx_base_port.h
/usr/include/bellagio/omx_base_sink.h
/usr/include/bellagio/omx_base_source.h
/usr/include/bellagio/omx_base_video_port.h
/usr/include/bellagio/omx_classmagic.h
/usr/include/bellagio/omx_comp_debug_levels.h
/usr/include/bellagio/omx_reference_resource_manager.h
/usr/include/bellagio/omxcore.h
/usr/include/bellagio/queue.h
/usr/include/bellagio/st_static_component_loader.h
/usr/include/bellagio/tsemaphore.h
/usr/include/bellagio/user_debug_levels.h
/usr/include/bellagio/utils.h
/usr/lib
/usr/lib/arm-linux-gnueabihf
/usr/lib/arm-linux-gnueabihf/libomxil-bellagio.a
/usr/lib/arm-linux-gnueabihf/pkgconfig
/usr/lib/arm-linux-gnueabihf/pkgconfig/libomxil-bellagio.pc
/usr/share
/usr/share/doc
/usr/share/doc/libomxil-bellagio-dev
/usr/share/doc/libomxil-bellagio-dev/changelog.Debian.gz
/usr/share/doc/libomxil-bellagio-dev/changelog.gz
/usr/share/doc/libomxil-bellagio-dev/copyright
/usr/lib/arm-linux-gnueabihf/libomxil-bellagio.so`

So I decided to include those directory to include files that --enable-omx-rpi required but it give me the same error again...

Here my command for the configuration:
`./configure --extra-cflags="-I/usr/local/include -I/usr/include" --extra-ldflags="-L/usr/local/lib -L/usr/lib" --extra-libs="-lpthread -lm -latomic" --arch=armel --enable-gmp --enable-gpl --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libfdk-aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libvorbis --enable-libvpx --enable-libzimg --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-mmal --enable-nonfree --enable-omx --enable-omx-rpi --enable-version3 --target-os=linux --enable-pthreads --enable-openssl --enable-hardcoded-tables`

Here the end of ffmpeg logs:
`BEGIN /tmp/ffconf.wV6otvMR/test.c
    1   #include <OMX_Core.h>
    2   int main(void) { OMX_IndexConfigBrcmVideoRequestIFrame; return 0; }
END /tmp/ffconf.wV6otvMR/test.c
gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -I/usr/local/include -I/usr/include -std=c11 -fomit-frame-pointer -marm -pthread -I/usr/local/include -I/usr/include/uuid -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/fribidi -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/local/include -I/usr/include/libdrm -I/usr/local/include -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/local/include -I/usr/include/opus -I/usr/include/opus -I/usr/include/p11-kit-1 -I/usr/local/include -I/usr/local/include -I/usr/local/include -I/usr/local/include -DX264_API_IMPORTS -I/usr/local/include -I/usr/include/libxml2 -c -o /tmp/ffconf.wV6otvMR/test.o /tmp/ffconf.wV6otvMR/test.c
/tmp/ffconf.wV6otvMR/test.c: In function 'main':
/tmp/ffconf.wV6otvMR/test.c:2:18: error: 'OMX_IndexConfigBrcmVideoRequestIFrame' undeclared (first use in this function); did you mean 'OMX_IndexConfigVideoFramerate'?
    2 | int main(void) { OMX_IndexConfigBrcmVideoRequestIFrame; return 0; }
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  OMX_IndexConfigVideoFramerate
/tmp/ffconf.wV6otvMR/test.c:2:18: note: each undeclared identifier is reported only once for each function it appears in
test_code cc OMX_Core.h OMX_IndexConfigBrcmVideoRequestIFrame
test_cc
BEGIN /tmp/ffconf.wV6otvMR/test.c
    1   #include <OMX_Core.h>
    2   int main(void) { OMX_IndexConfigBrcmVideoRequestIFrame; return 0; }
END /tmp/ffconf.wV6otvMR/test.c
gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -I/usr/local/include -I/usr/include -std=c11 -fomit-frame-pointer -marm -pthread -I/usr/local/include -I/usr/include/uuid -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/fribidi -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/local/include -I/usr/include/libdrm -I/usr/local/include -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/local/include -I/usr/include/opus -I/usr/include/opus -I/usr/include/p11-kit-1 -I/usr/local/include -I/usr/local/include -I/usr/local/include -I/usr/local/include -DX264_API_IMPORTS -I/usr/local/include -I/usr/include/libxml2 -isystem/opt/vc/include/IL -c -o /tmp/ffconf.wV6otvMR/test.o /tmp/ffconf.wV6otvMR/test.c
/tmp/ffconf.wV6otvMR/test.c: In function 'main':
/tmp/ffconf.wV6otvMR/test.c:2:18: error: 'OMX_IndexConfigBrcmVideoRequestIFrame' undeclared (first use in this function); did you mean 'OMX_IndexConfigVideoFramerate'?
    2 | int main(void) { OMX_IndexConfigBrcmVideoRequestIFrame; return 0; }
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  OMX_IndexConfigVideoFramerate
/tmp/ffconf.wV6otvMR/test.c:2:18: note: each undeclared identifier is reported only once for each function it appears in
ERROR: OpenMAX IL headers from raspberrypi/firmware not found`

What I'm doing wrong? How can I solve this problem?

6by9, 2022-01-26:
OpenMax is not supported on the 64 bit OS.

closed 2022-01-26 by pelwell

dbuscaino, 2022-01-26:
There is a way to have hardware acceleration with ffmpeg?

6by9, 2022-01-26:
> There is a way to have hardware acceleration with ffmpeg?

v4l2m2m.

dbuscaino, 2022-01-26:
> v4l2m2m.

Can you tell me how can I include it on ffmpeg configuration?

6by9, 2022-01-26:
It should auto detect it. [Configure](https://github.com/FFmpeg/FFmpeg/blob/master/configure#L349) only has:
```
--disable-v4l2-m2m   disable V4L2 mem2mem code [autodetect]
```

```
pi@raspberrypi:~ $ ffmpeg -decoders | grep m2m
...
 V..... h263_v4l2m2m         V4L2 mem2mem H.263 decoder wrapper (codec h263)
 V..... h264_v4l2m2m         V4L2 mem2mem H.264 decoder wrapper (codec h264)
 V..... hevc_v4l2m2m         V4L2 mem2mem HEVC decoder wrapper (codec hevc)
 V..... mpeg1_v4l2m2m        V4L2 mem2mem MPEG1 decoder wrapper (codec mpeg1video)
 V..... mpeg2_v4l2m2m        V4L2 mem2mem MPEG2 decoder wrapper (codec mpeg2video)
 V..... mpeg4_v4l2m2m        V4L2 mem2mem MPEG4 decoder wrapper (codec mpeg4)
 V..... vc1_v4l2m2m          V4L2 mem2mem VC1 decoder wrapper (codec vc1)
 V..... vp8_v4l2m2m          V4L2 mem2mem VP8 decoder wrapper (codec vp8)
 V..... vp9_v4l2m2m          V4L2 mem2mem VP9 decoder wrapper (codec vp9)
pi@raspberrypi:~ $ ffmpeg -encoders | grep m2m
...
 V..... h263_v4l2m2m         V4L2 mem2mem H.263 encoder wrapper (codec h263)
 V..... h264_v4l2m2m         V4L2 mem2mem H.264 encoder wrapper (codec h264)
 V..... hevc_v4l2m2m         V4L2 mem2mem HEVC encoder wrapper (codec hevc)
 V..... mpeg4_v4l2m2m        V4L2 mem2mem MPEG4 encoder wrapper (codec mpeg4)
 V..... vp8_v4l2m2m          V4L2 mem2mem VP8 encoder wrapper (codec vp8)
```
Note that FFmpeg does not filter for the decoders that are actually available, only those that support has been compiled in for.

dbuscaino, 2022-01-26:
> It should auto detect it. [Configure](https://github.com/FFmpeg/FFmpeg/blob/master/configure#L349) only has:
>
> ```
> --disable-v4l2-m2m   disable V4L2 mem2mem code [autodetect]
> ```
>
> ``` [...]

Thank you so much! I saw --disable-v4l2-m2m option but I don't know if that meas it is enable by default! Thank you again!

lurch, 2022-01-27:
You might want to suggest to the author of the original instructions you were following, that they add some updated info? :slightly_smiling_face:

dbuscaino, 2022-01-27:
> You might want to suggest to the author of the original instructions you were following, that they add some updated info? slightly_smiling_face

Yes sure! Thanks for your suggestion! slightly_smiling_face