raspberrypi/Raspberry-Pi-OS-64bit #204: Is it possible to migrate from the beta to the release version?

jiriks74, 2022-02-05:
Hi,
I've been using beta 64bit Pi OS for 2 years now (server), and since there's now an official release I wonder if it's possible to migrate to it (the beta one is basically Debian as far as I know) without reinstalling the whole thing.

Thanks for developing that awesome thing btw

lurch, 2022-02-05:
Answered here: https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/#comment-1580125
But of course there's no guarantees that it'll work if your original-install was from 2 years ago, so always a good idea to make a backup just to be on the safe side.

jiriks74, 2022-02-05:
> Answered here: https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/#comment-1580125 But of course there's no guarantees that it'll work if your original-install was from 2 years ago, so always a good idea to make a backup just to be on the safe side.

Yes, I'm aware that it would not be officially supported and I won't blame u for me messing it up. Thanks for your help.

jiriks74, 2022-02-23:
Looks like everything's ok, thanks

closed 2022-02-23 by jiriks74