raspberrypi/Raspberry-Pi-OS-64bit #205: Merus sound hat does not work

twist-throttle, 2022-02-07:
I can't figure out how to get sound from the Merus hat, it was working fine on Buster. Below is the output from the aplay command. Any ideas?

 aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpimerusamp [snd_rpi_merus_amp], device 0: Merus Audio Amp ma120x0p-amp-0 [Merus Audio Amp ma120x0p-amp-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
sysdefault
    Default Audio Device
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
hw:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi1
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
hw:CARD=sndrpimerusamp,DEV=0
    snd_rpi_merus_amp, Merus Audio Amp ma120x0p-amp-0
    Direct hardware device without any conversions
plughw:CARD=sndrpimerusamp,DEV=0
    snd_rpi_merus_amp, Merus Audio Amp ma120x0p-amp-0
    Hardware device with all software conversions
sysdefault:CARD=sndrpimerusamp
    snd_rpi_merus_amp, Merus Audio Amp ma120x0p-amp-0
    Default Audio Device
dmix:CARD=sndrpimerusamp,DEV=0
    snd_rpi_merus_amp, Merus Audio Amp ma120x0p-amp-0
    Direct sample mixing device
usbstream:CARD=sndrpimerusamp
    snd_rpi_merus_amp
    USB Stream Output

pelwell, 2022-02-07:
Have you tried right-clicking the volume control to select the sound output?

twist-throttle, 2022-02-08:
I tried that right away.

The hat kinda works but super scratchy sounding. It took reinstalling Bullseye 64bit to my SSD and modifying config.txt BEFORE booting the first time. The modifications are from the Merus instructions. 

This is different than when I was on  a 32 bit OS.

Does anyone know what setting needs adjustment so things sound correct and not unintelligible scratchy noise?

twist-throttle, 2022-02-09:
I got things working though not sure exactly what config.txt setting did it. The problem appeared to be the video driver conflicting with the Merus HAT.

This is a Pi4 4Gig, with Sunfounder 7" HDMI touch display, and Infineon Merus HAT driving a single speaker wired in parallel bridged mode for higher output, and a Geekworm UPS HAT V3.

Below is my config.txt with comments removed:
dtparam=i2c_arm=on
dtparam=i2s=on

dtparam=audio=off
dtoverlay=merus-amp
gpio=8=op,dl
#The above enables parallel bridge mode for audio

display_auto_detect=1

dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2

arm_64bit=1

disable_overscan=1

[cm4]
otg_mode=1

[pi4]
arm_boost=1