raspberrypi/Raspberry-Pi-OS-64bit #206: pager issue

forthprgrmr, 2022-02-07:
Enabling pager brings up multiple boxes on right edge of Task Bar, but they don't work.
Get brown screens on each desktop and can't do anything including going back to (1) - it's all brown now too
must reboot to use system

frozenfoxx, 2022-02-16:
Same problem here. I've found that it looks like it has something to do with the window manager crashing but it's hard to diagnose when the desktop completely dies.

lurch, 2022-02-21:
Probably the same issue as described [here](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/#comment-1580177) ?

frozenfoxx, 2022-02-25:
Yes, it's apparently got something to do with Mutter.

lurch, 2022-02-25:
ping @spl237  (although I've got a vague memory of somebody saying that virtual desktops aren't officially supported? :shrug: )

spl237, 2022-02-25:
No, we don't support virtual desktops - we never have. Mutter seems to have broken them completely, I'm afraid.