raspberrypi/Raspberry-Pi-OS-64bit #208: Last update have broken chromium

Maxim766, 2022-02-12:
After last OS update chromium browser stopped launching youtube videos, It's says to me, that it's audio issue and I need to reboot my device (doesn't help). Updated through Updater, not by terminal with "sudo apt full-upgrade"

Maxim766, 2022-02-19:
Well, firefox and vivaldi works fine with youtube videos, It seems only chromium has that problem (still waitng for fix)

XECDesign, 2022-02-21:
Any steps on how to reproduce the issue? (Starting with an Raspberry Pi OS Bullseye install?)

Maxim766, 2022-02-21:
0) Have installed Raspberry Pi OS Bullseye
1) Made update through Updater (applet at the right bottom)

Maxim766, 2022-02-21:
> @Maxim766 Did you miss out "0.5. Move task bar to the bottom"?

heh, yes)

Maxim766, 2022-02-21:
ok, let me just find out what is it, maybe It's just broken libraries

Maxim766, 2022-02-21:
Ok, I found it. It's because of openjdk-8-jdk unstable (sid) libraries, using old libs (stretch) doesn't matters on browser. 

But I'm still curious, why vivaldi and firefox are working fine with newer unstable libs?