raspberrypi/Raspberry-Pi-OS-64bit #210: miniupnpd on Raspberry PI OS Bullseye seems broken

woehrl, 2022-02-15:
Hi!
nftables seems to be the iptables alternative used within Raspberry PI OS Bullseye x64.
miniupnpd relies on the script `/usr/libexec/miniupnpd-startstop-helper.sh` to determine the correct ip filter implementation to use, either iptables or nftables.

The error seems to be in the "check_miniupnpd()" method:
```
system_backend=$(update-alternatives --query iptables | grep -F Value | grep -qF nft \
                && echo nftables || echo iptables)
```

This query falsely returns "iptables" because `update-alternatives --query iptables` returns the error message `update-alternatives: error: no alternatives for iptables`. So nftables here is no valid alternative for iptables.

Imho the Fix would be to either have a different approach to detect the correct library to use or to set nftables as a valid alternative to iptables.

Best regards,
 Florian

lurch, 2022-02-15:
Is this a Raspberry-Pi-OS-specific problem, or does it affect upstream Debian Bullseye (e.g. on a desktop PC) too?

XECDesign, 2022-02-16:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=992603

Not sure why they didn't add the fix to the Bullseye version of the package, but I've backported it anyway Hopefully the problem should be resolved.

woehrl, 2022-02-16:
I thought it's maybe a Raspberry PI OS specific problem of the package definition (nft as alternative to iptables) but it was indeed a debian bug, sorry for posting here. And thx for the quick response and fix!

closed 2022-02-16 by woehrl