raspberrypi/Raspberry-Pi-OS-64bit #211: Icon disappears from desktop when dragging it to another spot

fieldbird, 2022-02-22:
Icon reappears after logging out then logging back in

lurch, 2022-02-22:
Sounds like one for @spl237 ?

@fieldbird Is this unique to the 64-bit Raspberry Pi OS, or does the same thing happen on the 32-bit Raspberry Pi OS as well?

fieldbird, 2022-02-23:
Happens only on 64-bit Raspberry Pi OS.
32-bit Raspberry Pi OS works fine.

spl237, 2022-02-23:
Actually, I've tried it on 32-bit, and it seems to happen on there too - I'll see if I can see what is happening.

spl237, 2022-02-23:
Ah - no, it's not quite the same problem. Under 32-bit, the icon disappears when you stop dragging, but immediately re-appears as soon as you click the desktop; no need to log out and back in.

fieldbird, 2022-02-23:
Inadvertently clicked "Close with comment" instead of "Comment"

spl237, 2022-03-11:
I think I've got a fix for this - it certainly fixes the similar issue on the 32 bit version. It'll be in an upcoming release.