raspberrypi/Raspberry-Pi-OS-64bit #212: Pixel Doubling Bricks the Raspberry Pi

HoldOffHunger, 2022-02-27:
Hi, all,

If you neable "Pixel-Doubling" from within "Display Options" of `raspi-config` and reboot, using a Raspberry Pi 3b+, you'll brick your device and won't be able to log in.   There is a greater explanation here of the technical details why this is failing: https://raspberrypi.stackexchange.com/questions/133807/not-booting-after-enabling-pixel-doubling

Cheers,

pelwell, 2022-02-28:
I can confirm the issue with a 4B:
```
Feb 28 09:10:59 raspberrypi systemd[1]: Starting Light Display Manager...
Feb 28 09:10:59 raspberrypi lightdm[1385]: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by any .service files
Feb 28 09:10:59 raspberrypi systemd[1]: Started Light Display Manager.
Feb 28 09:11:01 raspberrypi lightdm[1397]: xrandr: unrecognized option '--filter'
Feb 28 09:11:01 raspberrypi lightdm[1397]: Try 'xrandr --help' for more information.
Feb 28 09:11:03 raspberrypi lightdm[1405]: xrandr: unrecognized option '--filter'
Feb 28 09:11:03 raspberrypi lightdm[1405]: Try 'xrandr --help' for more information.
Feb 28 09:11:04 raspberrypi systemd[1]: lightdm.service: Main process exited, code=exited, status=1/FAILURE
Feb 28 09:11:04 raspberrypi systemd[1]: lightdm.service: Failed with result 'exit-code'.
Feb 28 09:11:04 raspberrypi systemd[1]: lightdm.service: Consumed 1.889s CPU time.
Feb 28 09:11:04 raspberrypi systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 25.
Feb 28 09:11:04 raspberrypi systemd[1]: Stopped Light Display Manager.
```

pelwell, 2022-02-28:
This issue really belongs in the raspi-config repo: https://github.com/RPi-Distro/raspi-config
I would move the issue for you, but for historical reasons it is under the RPi-Distro organisation (also ours), and GitHub doesn't support moving issues between organisations.

XECDesign, 2022-02-28:
@spl237

spl237, 2022-02-28:
As it happens, I am currently in the process of removing this option - it's not needed under KMS, where you can just use the screen resolution tool to set a lower resolution.

HoldOffHunger, 2022-03-01:
Excellent, thanks all for taking a look at this.  I'll forward any future issues with `raspi-config` to the above-linked github, thanks.