raspberrypi/Raspberry-Pi-OS-64bit #213: MIPI DSI + HDMI Wallpaper/Desktop Color Problem

derlucas, 2022-03-06:
Having a Model 4B (tested with two pi's). I installed plain Raspberry PI OS 64Bit with Desktop Environment.

When booting with the original 7" Touch Display only, i can use it properly, touch is working fine and i can change background of lxde with the "Appearance Settings" app to either a Wallpaper Picture or solid color.

Then connecting a HDMI display, the desktop expands onto the additional display. However the background is just a green-ish color gradient. I tried to change the background to solid color with the Appearance Settings program on the "Desktop HDMI-2" tab without any success. I can move windows onto the second desktop, play Videos with VLC without issues.

It does not matter if the HDMI display is connected before powering up the PI or during power-on after boot.

I plugged the HDMI cable into the port further away from the USB-C to make it not the primary display. However the issues described appear regardless of which HDMI port is used.

For testing and creating a proper screnshot i used a USB-HDMI capture stick. In the picture you see the desktop of the HDMI port with the File Manager Window moved to side and the Wallpaper which i cannot change.
[image]

When i kill the pcmanfm process and start it again with "pcmanfm --desktop", then the black background is set correctly,

lurch, 2022-03-10:
ping @spl237 and @6by9

ubopod, 2023-03-10:
I am seeing the same problem in 32 bit OS after upgrade. Setting wallpaper image does not change anything...