raspberrypi/Raspberry-Pi-OS-64bit #215: Sudo Raspi-Config

carnby77, 2022-03-07:
Changing to predicatble names for eth0 does not work, 
When running ifconfig command it still shows as eth0 instead of enxe45f01964XXX.
Is the raspi-config unreliable?

XECDesign, 2022-03-07:
I don't remember the details, but the short version is that's just how the rules work. IIRC, you'll only see predictable interface names apply to devices on PCI and USB buses. I believe that means that devices where ethernet hangs off USB will use them, but ones where they are on their own MDIO bus (pi4 and newer), will not.

It's more of an upstream issue. Although we could fix it downstream, I suspect that there will be more complaints that we're changing the behaviour and breaking some setups. Maybe with the next major Debian release.

rkarlsba, 2022-03-31:
You could do it manually, though, with udev: https://wiki.debian.org/NetworkInterfaceNames

carnby77, 2022-03-31:
Thank you very much for your reply. I will read through it.