raspberrypi/Raspberry-Pi-OS-64bit #216: uvcdynctrl is missing

soyersoyer, 2022-03-16:
The uvcdynctrl packages are included in the 32-bit Bullseye, but not in the 64-bit one.

uvcdynctrl - Command line tool to control v4l2 devices
uvcdynctrl-data - Command line tool to control v4l2 devices - data files
uvcdynctrl-dbg - Debug Symbols for uvcdynctrl

lurch, 2022-03-17:
https://packages.debian.org/search?keywords=uvcdynctrl suggests that the package is only available in Buster, and not available in Bullseye? 
Both the 32-bit and 64-bit versions of [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/) are now based on Bullseye, so if this package is critical to you I guess you'll need to use "Raspberry Pi OS (Legacy)" ?

soyersoyer, 2022-03-17:
I can install it on the 32-bit version of Raspberry Pi OS 11 (Bullseye).
The [guvcview package](https://packages.debian.org/bullseye/guvcview) recommends it on Debian Bullseye too.
I think there shouldn't be differences in the package list.

lurch, 2022-03-17:
ping @XECDesign

XECDesign, 2022-03-17:
As @lurch said, the package is not in Debian.

Raspbian seems to copy all packages across to newer releases, without monitoring for removal, which is why you're seeing it in Raspberry Pi OS for armhf, but not for arm64 (which is based on Debian instead).

Since the package is not one of ours and isn't in any of out repos, it should be reported to Raspbian: https://bugs.launchpad.net/raspbian