raspberrypi/Raspberry-Pi-OS-64bit #218: Rasperry Pi OS should keep previous kernel

rkarlsba, 2022-03-31:
I just reported #217 and it seems to me that rpios is a bit too conservative on the disk usage. Currently, there doesn't seem to be a way to roll back to the previous kernel if the midden hits the windmill, so to say. It should be prominent to have a boot partition large enough to allow for at least two kernel versions in case the new one just doesn't work.

lurch, 2022-03-31:
> in case the new one just doesn't work

Mitigating against this, each new kernel-release goes through extensive testing before being pushed into the `apt` repo (that's what the use-at-your-own-risk `rpi-update` is [all about](https://www.raspberrypi.com/documentation/computers/os.html#rpi-update)), and the Raspberry Pi is fairly "fixed" from a hardware point of view.

XECDesign, 2022-03-31:
The boot partition should currently be large enough. It's just that our kernel package isn't based on the one provided by Debian and isn't set up in a way that allows for multiple kernels.

But yes, ideally we should have a way to recover. I just can't think of one that doesn't involve taking the card out and manually fixing things on another computer, even if we do keep old kernels around.

> Mitigating against this, each new kernel-release goes through extensive testing before being pushed into the `apt` repo (that's what the use-at-your-own-risk `rpi-update` is [all [...]

Things do slip through though. The linked issue is not something that would've been easy to spot in testing.

lurch, 2022-03-31:
> Things do slip through though. The linked issue is not something that would've been easy to spot in testing.

Yeah, but arguably "wifi not connecting" is something quite different to "the kernel not booting at all" :slightly_smiling_face: 
(although admittedly there's not much difference to the end-user if you're only connecting wirelessly to a headless Raspberry Pi)

rkarlsba, 2022-03-31:
Appartently, #217 may not have been a kernel bug, but a bug in a dependency (regulatory.db). It still broke wifi, though (see the bug for details), so having the chance to just like in grub choose "last well know" or something would bevery practical

lurch, 2022-03-31:
> just like in grub choose "last well know"

On the RPi that might be possible with something like UBoot (?), but the way that Raspberry Pi OS works, the Linux kernel is the first thing that gets loaded by `start.elf`, so there's no mechanism to interactively select which kernel gets used. (more info [here](https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained) if you're curious, although note that on a Pi4 the job of `bootcode.bin` is instead done by the EEPROM bootloader)