raspberrypi/Raspberry-Pi-OS-64bit #219: Raspberry Pi 4 Kodi Bluetooth speaker disconnection problem

suqin-haha, 2022-04-03:
Hi,

Here is new problem that I cannot use Bluetooth speaker with Kodi, the Bluetooth speaker is good for others.

My observation:
the Bluetooth speaker will loss connection when I open Kodi... then there is no sound.
However, my Bluetooth mouse doesn't loss connection (Or it reconnects too fast to notice).
And after I quit Kodi, if I click the tray area, like the volume button or the Bluetooth button, the tray area will refresh and reconnect the Bluetooth device.

Log file:
[kodi.log](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8403485/kodi.log)

Environment is same, still raspberry pi 4 with 64bit Pi OS, and as I remember it works no problem on 32bit.
one of test file I use is: [jellyfish-55-mbps-hd-h264.mkv](https://jell.yfish.us/media/jellyfish-55-mbps-hd-h264.mkv)

Thanks!

suqin-haha, 2022-04-25:
More info:
tried to reconnect my Bluetooth speaker by terminal while Kodi opening.
just cannot connect: 
```
[bluetooth]# pair FC:A8:9A:27:8D:3A
Attempting to pair with FC:A8:9A:27:8D:3A
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# connect FC:A8:9A:27:8D:3A
Attempting to connect to FC:A8:9A:27:8D:3A
Failed to connect: org.bluez.Error.Failed
```
even remove and pair again didn't work
```
[bluetooth]# pair FC:A8:9A:27:8D:3A
Attempting to pair with FC:A8:9A:27:8D:3A
[CHG] Device FC:A8:9A:27:8D:3A Connected: yes
[CHG] Device FC:A8:9A:27:8D:3A Modalias: bluetooth:v0039p1582d2212
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device FC:A8:9A:27:8D:3A ServicesResolved: yes
[CHG] Device FC:A8:9A:27:8D:3A Paired: yes
Pairing successful
[CHG] Device FC:A8:9A:27:8D:3A ServicesResolved: no
[CHG] Device FC:A8:9A:27:8D:3A Connected: no

```

my speaker info
```
[bluetooth]# info FC:A8:9A:27:8D:3A
Device FC:A8:9A:27:8D:3A (public)
        Name: SRS-XB2
        Alias: SRS-XB2
        Class: 0x00240414
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v0039p1582d2212
        ManufacturerData Key: 0x5349
        ManufacturerData Value:
  53 43                                            SC

```

Gex-dev69, 2022-10-01:
Did you find solution on other forums?

popcornmix, 2022-10-01:
Does launching `kodi --pulse` help?

suqin-haha, 2022-10-02:
> Does launching `kodi --pulse` help?

yes! that works! Why is that happens, Is there any config or settings I can do?

Gex-dev69, 2022-10-02:
> Does launching `kodi --pulse` help?

Bless.

popcornmix, 2022-10-02:
> BTW, Why is that happens, Is there any config or settings I can do?

pulse prevents advanced alsa features like passthough audio which can be important for kodi users,
so the launcher script disables pulse unless `--pulse` is present as a command line option, or `WANT_PULSE` is defined as an env variable.

If you want to pass this option from desktop launcher you can edit file shown [here](https://forums.raspberrypi.com/viewtopic.php?p=2039334#p2039334)

suqin-haha, 2022-10-03:
> pulse prevents advanced alsa features like passthough audio which can be important for kodi users, so the launcher script disables pulse unless `--pulse` is present as a command line option, or [...]

Great! Thanks!

closed 2022-10-03 by suqin-haha