raspberrypi/Raspberry-Pi-OS-64bit #221: Wi-Fi not supported on Channel 13 in Australia with RTL8188EUS (0bda:8179)

ITmaze, 2022-04-13:
After noticing that my headless pi was no longer connecting to the network, I discovered that my USB dongle can only see channels 1-11. I note that I also have the same message `cfg80211: loaded regulatory.db is malformed or signature is missing/invalid` in my logs as seen in issue #217.

Note that this hardware has previously worked, albeit on channel 1 and without setting the country code.

During testing I managed to set my country code to Australia, which took some extra doing, forcing it in two places, `/etc/modprobe.d/cfg80211.conf` and `/etc/default/crda`

```
# iw reg get 
global
country AU: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 36), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5600 @ 80), (N/A, 27), (0 ms), DFS
	(5650 - 5730 @ 80), (N/A, 27), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 36), (N/A)
	(57000 - 66000 @ 2160), (N/A, 43), (N/A), NO-OUTDOOR
```

I can only see the first 11 2.4 GHz channels:
```
# iwlist chan
lo        no frequency information.

eth0      no frequency information.

wlan0     11 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Current Frequency:2.412 GHz (Channel 1)

docker0   no frequency information.
```

I'm running the latest OS:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

After attempting to update using the normal `apt update && apt -y upgrade`, I finally updated with `rpi-update` which I understood would upate the Pi firmware (as recommended in Issue #217), which didn't fix the issue (the output below is the second run, the first run updated the firmware and kernel modules).
```
# rpi-update 
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date (delete /boot/.firmware_revision to force an update anyway)
```

I also tried forcing the reinstallation of the firmware: `apt --reinstall install firmware-brcm80211`

No doubt I'm missing something obvious, but my 23 years using debian hasn't been able to help this time ;-(

I'm running this pi hardware:
```
# awk '/^Revision/ {sub("^1000", "", $3); print $3}' /proc/cpuinfo
a21041
```

Which according to perturb.org/rpi?rev=a21041 is this:
```
Revision: a21041
Model: 2 Model B
Memory: 1 GB
Overvolt: No
Release: Q1 2015
Notes: (Mfg by Embest)
```

The wireless network hardware is:
```
# lsusb -vd 0bda:8179

Bus 001 Device 005: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8179 RTL8188EUS 802.11n Wireless Network Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 00E04C0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

lurch, 2022-04-13:
@ghollingworth I guess this ought to be transferred to the linux repo?

pelwell, 2022-04-13:
Have you tried the suggestion I made on the other issue?:

1. Download https://mirrors.edge.kernel.org/pub/software/network/wireless-regdb/wireless-regdb-2022.02.18.tar.xz
2. Extract `regulatory.db` and `regulatory.db.p7s`.
3. Copy them to `/lib/firmware/`.
4. Reboot.