raspberrypi/Raspberry-Pi-OS-64bit #222: Error when using Queue on Raspberry Pi

KilimSerg, 2022-04-15:
Hello!
Help me figure out the problem, I'm trying to transfer a python script from Windows 7 PyCharm to Raspberry Pi 4 8GB.
The script uses the multiprocessing module.

from multiprocessing import Process, cpu_count, Queue, Pipe

Process, cpu_count, Pipe - work fine with them there are no problems.
But there was a problem with Queue.

Queue is declared in a block:

if __name__ == '__main__':
qout = Queue()
------------------
------------------
procs=[]
    proc = Process(target=checkGen, args=(baseName, qout, profit, child_conn))
    procs.append(proc)
    proc.start()

multiprocessingCount = cpu_count()
for u in range(multiprocessingCount): there is not enough RAM, you need to reduce the number of threads
        proc = Process(target=genHashes, args=(key_count, qout))
        procs.append(proc)
        proc.start()

I will not give the entire code of the program so as not to clutter up the text.

Queue is used to pass data from the genHashes processes to the checkGen process.
The Raspberry Pi 4 8GB is running the Raspbian Buster Lite distribution upgraded to 64-bit.
Version Python 3.7.3
When running the script, it gives an error

Traceback (most recent call last):
  File "start_v7.py", line 241, in <module>
    qout = Queue()
  File "/usr/lib/python3.7/multiprocessing/context.py", line 102, in Queue
    return Queue(maxsize, ctx=self.get_context())
  File "/usr/lib/python3.7/multiprocessing/queues.py", line 42, in __init__
    self._rlock = ctx.Lock()
  File "/usr/lib/python3.7/multiprocessing/context.py", line 67, in Lock
    return Lock(ctx=self.get_context())
  File "/usr/lib/python3.7/multiprocessing/synchronize.py", line 162, in __init__
    SemLock.__init__(self, SEMAPHORE, 1, 1, ctx=ctx)
  File "/usr/lib/python3.7/multiprocessing/synchronize.py", line 59, in __init__
    unlink_now)
PermissionError: [Errno 13] Permission denied

Since I don't have a lot of experience with Raspberry Pi programming in python, I can't figure out what it is missing.
Maybe something else needs to be delivered, maybe something else.
If you can advise.
Sergey.

lurch, 2022-04-15:
This sounds very much like a question for the [forums](https://forums.raspberrypi.com/), rather than the Raspberry Pi OS 64-bit issue tracker.

KilimSerg, 2022-04-18:
The whole reason turned out to be that initially I used my raspberry to run the xmrig miner and for this I downloaded the minimum Lite 32-bit operating system there, upgrading it to 64 bits and installing xmrig.
Malinka's operating system originally contained python version 2.7, so I installed python version 3.7 on it. But it seems that during installation something became incorrect and therefore this one gave an error.
Now I have officially downloaded the 64-bit Lite operating system on which Python version 3.9 is installed and the script on it started without errors.

closed 2022-04-18 by KilimSerg