raspberrypi/Raspberry-Pi-OS-64bit #224: ffplay playing audio results in Segmentation fault

Andrewiski, 2022-04-26:
This works
```
aplay /usr/share/sounds/alsa/Front_Left.wav

```
Playing WAVE '/usr/share/sounds/alsa/Front_Left.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

This Fails with segmentation Fault

```
ffplay /usr/share/sounds/alsa/Front_Left.wav
```
ffplay version 4.3.4-0+deb11u1 Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --prefix=/usr --extra-version=0+deb11u1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Segmentation fault

uname -a
Linux deradarpi1 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

in config.txt 

```
 cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
enable_uart=1
```

Andrewiski, 2022-04-27:
installing with apt-get install ffmpeg causes Segmentation fault but a build from source works correctly

I did a rebuild of ffmpeg 4.3 and that fixed it but had to do it on a 8Gb model as the 1GB Raspberry Pi 4 Crashes with out of memory.

```

echo This Will Fail on 1GB Memory version of Raspberry PI 4 Use "top" to check ram size   
echo Downloading from Git hub https://github.com/FFmpeg/FFmpeg.git
cd ~
git clone --depth 1 --branch release/4.3 https://github.com/FFmpeg/FFmpeg.git
cd FFmpeg
sudo apt-get update -qq && sudo apt-get -y install autoconf automake build-essential cmake git-core libass-dev libfreetype6-dev libgnutls28-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo wget yasm zlib1g-dev libunistring-dev libdrm-dev libopus-dev libvpx-dev libwebp-dev libx264-dev libxml2-dev libfdk-aac-dev libmp3lame-dev

./configure --prefix=`pwd`/install \
    --enable-gpl \
    --enable-nonfree \
    --arch=aarch64 \
    --disable-libaom \
    --enable-libass \
    --enable-libfdk-aac \
    --enable-libfreetype \
    --enable-libmp3lame \
    --enable-libopus \
    --enable-libvorbis \
    --enable-libvpx \
    --enable-libx264 \
    --enable-libx265 \
    --enable-libxml2 \
    --enable-libwebp \
    --enable-libdrm

   make -j4
   make install

```

popcornmix, 2022-04-27:
> I did a rebuild of ffmpeg 4.3 and that fixed it but had to do it on a 8Gb model as the 1GB Raspberry Pi 4 Crashes with out of memory.
>   make -j4

A smaller j value would likely have worked on a Pi4. Increasing swap should also work.

@XECDesign with both repos updated:
On 32-bit:
```
ffplay version 4.3.3-0+rpt3+deb11u1 Copyright (c) 2003-2021 the FFmpeg developers
built with gcc 10 (Raspbian 10.2.1-6+rpi1)
```
On 64-bit:
```
ffplay version 4.3.4-0+deb11u1 Copyright (c) 2003-2021 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
```
Is the 64-bit ffplay built from @jc-kynesim's repo, or is it built from vanilla upstream debian ffmpeg?

XECDesign, 2022-04-27:
I think there was recently an embargoed ffmpeg update, which I haven't had a chance to add yet. Since Raspbian lags behind, it hasn't received it yet. Updated version should be up later today.

XECDesign, 2022-05-03:
The packages were updated last week. Did that help at all with the issue, or are you still seeing the segfault?

femajocri, 2023-09-23:
for me it only worked like this:

```
sudo apt-get update -qq && sudo apt-get -y install autoconf automake build-essential cmake git-core libaom-dev libass-dev libfreetype6-dev libgnutls28-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo wget yasm zlib1g-dev libunistring-dev libdrm-dev libopus-dev libvpx-dev libwebp-dev libx264-dev libx265-dev libxml2-dev libmp3lame-dev

mkdir /tmp/build && cd /tmp/build

wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/aac-enc_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac-dev_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac2_2.0.1-1_armhf.deb

sudo dpkg -i libfdk-aac2_2.0.1-1_armhf.deb libfdk-aac-dev_2.0.1-1_armhf.deb aac-enc_2.0.1-1_armhf.deb

cd ~
git clone --depth 1 --branch release/6.0 https://github.com/FFmpeg/FFmpeg.git
cd FFmpeg

./configure --prefix=`pwd`/install \
    --enable-gpl \
    --enable-nonfree \
    --arch=aarch64 \
    --enable-libaom \
    --enable-libass \
    --enable-libfdk-aac \
    --enable-libfreetype \
    --enable-libmp3lame \
    --enable-libopus \
    --enable-libvorbis \
    --enable-libvpx \
    --enable-libx264 \
    --enable-libx265 \
    --enable-libxml2 \
    --enable-libwebp \
    --enable-libdrm

make -j4
make install
```

femajocri, 2023-09-23:
for me it only worked like this:

```
sudo apt-get update -qq && sudo apt-get -y install autoconf automake build-essential cmake git-core libaom-dev libass-dev libfreetype6-dev libgnutls28-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo wget yasm zlib1g-dev libunistring-dev libdrm-dev libopus-dev libvpx-dev libwebp-dev libx264-dev libx265-dev libxml2-dev libmp3lame-dev

mkdir /tmp/build && cd /tmp/build

wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/aac-enc_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac-dev_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac2_2.0.1-1_armhf.deb

sudo dpkg -i libfdk-aac2_2.0.1-1_armhf.deb libfdk-aac-dev_2.0.1-1_armhf.deb aac-enc_2.0.1-1_armhf.deb

cd ~
git clone --depth 1 --branch release/6.0 https://github.com/FFmpeg/FFmpeg.git
cd FFmpeg

./configure --prefix=`pwd`/install \
    --enable-gpl \
    --enable-nonfree \
    --arch=arm \
    --enable-libaom \
    --enable-libass \
    --enable-libfdk-aac \
    --enable-libfreetype \
    --enable-libmp3lame \
    --enable-libopus \
    --enable-libvorbis \
    --enable-libvpx \
    --enable-libx264 \
    --enable-libx265 \
    --enable-libxml2 \
    --enable-libwebp \
    --enable-libdrm

make -j4
make install
```

Had to remove libfdk-aac from apt-get (not found),
had to get the download libfdk-aac libraries,
and had to change arch to arm.

femajocri, 2023-09-23:
for me it only worked like this:

```
sudo apt-get update -qq && sudo apt-get -y install autoconf automake build-essential cmake git-core libaom-dev libass-dev libfreetype6-dev libgnutls28-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev pkg-config texinfo wget yasm zlib1g-dev libunistring-dev libdrm-dev libopus-dev libvpx-dev libwebp-dev libx264-dev libx265-dev libxml2-dev libmp3lame-dev

mkdir /tmp/build && cd /tmp/build

wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/aac-enc_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac-dev_2.0.1-1_armhf.deb
wget http://ftp.de.debian.org/debian/pool/non-free/f/fdk-aac/libfdk-aac2_2.0.1-1_armhf.deb

sudo dpkg -i libfdk-aac2_2.0.1-1_armhf.deb libfdk-aac-dev_2.0.1-1_armhf.deb aac-enc_2.0.1-1_armhf.deb

cd ~
git clone --depth 1 --branch release/6.0 https://github.com/FFmpeg/FFmpeg.git
cd FFmpeg

./configure --prefix=`pwd`/install \
    --enable-gpl \
    --enable-nonfree \
    --arch=arm \
    --enable-libaom \
    --enable-libass \
    --enable-libfdk-aac \
    --enable-libfreetype \
    --enable-libmp3lame \
    --enable-libopus \
    --enable-libvorbis \
    --enable-libvpx \
    --enable-libx264 \
    --enable-libx265 \
    --enable-libxml2 \
    --enable-libwebp \
    --enable-libdrm

make -j4
make install
```

Had to remove libfdk-aac from apt-get (not found),
had to install the download libfdk-aac libraries,
and had to change arch to arm on configure options.

closed 2023-09-25 by XECDesign