raspberrypi/Raspberry-Pi-OS-64bit #225: Maximized window does not fill Desktop in Headless Config VNC server 1080p resolution Pi OS 64bit

mv4manishverma, 2022-05-06:
Using "Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux"

OS: Pi OS Bullseye Release 11.

I am using CM4 ioboard carrier board with 4GB RAM and 32GB eMMC compute module 4.
I set the headless resolution to 1920x1080 from the Menu while on the external monitor connected to HDMI.
After that I removed the external monitor and rebooted the board. Then I connected to the board from VNC viewer. The Desktop appeared in 1920x1080 resolution but if I open any window or terminal and try to maximize it, it does not fill the whole desktop.

lurch, 2022-05-07:
Could you attach a screenshot / photo to show what you mean please?

XECDesign, 2022-05-09:
@spl237 sounds like openbox is not reading the root window size correctly in that situation?

mv4manishverma, 2022-05-09:
> 
[image]
[image]

Here are the screen shots.

mv4manishverma, 2022-05-09:
> @spl237 sounds like openbox is not reading the root window size correctly in that situation?

Not sure what is the cause of this problem !

pelwell, 2022-05-09:
FYI the bottom right corner of the maximised window is at (1023,767).

lurch, 2022-05-09:
Thanks - based on your screenshot and @XECDesign 's comment, it seems that openbox (the window manager) thinks your screensize is 1024 x 768.

Hah, @pelwell _just_ beat me to it :laughing:

mv4manishverma, 2022-05-09:
> Thanks - based on your screenshot and @XECDesign 's comment, it seems that openbox (the window manager) thinks your screensize is 1024 x 768.
> 
> Hah, @pelwell _just_ beat me to it laughing

So what is the solution for this? Do i have to manually set the desired resolution?

spl237, 2022-05-09:
I've raised the issue with RealVNC - they will investigate.

mv4manishverma, 2022-05-09:
Thanks a lot !!

jmlineb, 2022-05-31:
I've had the same problem for a couple of months now. As a result, Raspberry Pi OS 64 bit is only usable for me with a monitor attached. I used it on a Raspberry Pi 4 with 4 GB of memory.

evansste12432, 2022-06-15:
My solution has been to set the resolution, on the raspberry pi, to match that which is used by VNC.

While you're viewing the Raspberry Pi through VNC, go through:

Start -> Preferences -> Raspberry Pi Configuration -> Display -> Headless Resolution.

Where "Start" is the raspberry icon, in the upper left corner.  Make sure you set this resolution so that it's the same that VNC is using.

I'm using tightvncserver.  So if I start tightvncserver with "tightvncserver -geometry 1280x720", at the command line, and then set the Raspberry Pi resolution, in the above path, to 1280x720, then full screen works properly.

evansste12432, 2022-06-16:
My solution has been to set the resolution, on the raspberry pi, to match that which is used by VNC.

While you're viewing the Raspberry Pi through VNC, go through:

Start -> Preferences -> Raspberry Pi Configuration -> Display -> Headless Resolution.

Where "Start" is the raspberry icon, in the upper left corner.  Make sure you set this resolution so that it's the same that VNC is using.

I'm using tightvncserver.  So if I start tightvncserver with "tightvncserver -geometry 1280x720", at the command line, and then set the Raspberry Pi resolution, in the above path, to 1280x720, then full screen has a chance to work properly.  These changes, only have to be made, once.

Once all of this is done, the order, in which, you choose to go "full screen", is important.  First, once you're viewing your Pi through VNC, you have to make VNC go full screen.  After that, you can then make your application, on the raspberry pi, go full screen.

If you don't make VNC go full screen, first, the application, on the raspberry pi, may not line up correctly, when you make that application go full screen.

Obviously, this isn't a perfect solution, but it does tend to work.  Hopefully, permanent fixes will soon be implemented by the 64-bit Raspberry Pi creators.

romton843, 2022-07-22:
Hi. I have the same issue. Some people claimed they solved it by modifying the DRM VC4 V3D driver lines in the config.txt file. It didn’t work for me though. Any other lead?

yvessa, 2022-07-22:
I'm using a HDMI dummy plug... not elegant but it's working!

jmlineb, 2022-07-22:
@romton843, I'm so sorry that the fix you mentioned is not working for you. But it gave me the idea to search the Web for that particular solution, and I found it, and it is working for me!  Here are the details, inspired both by your comment and by https://forums.raspberrypi.com/viewtopic.php?t=282380.

1. Comment out all dtoverlay=vc4-fkms-v3d lines in config.txt
2. Comment out max_framebuffers too (but this might not be necessary)
3. Restart the Pi headless
4. Connect with VNC client. Best not to maximize that VNC client but let it the max resolution determine the window size. Otherwise you get some mouse anomalies trying to switch from window to window.
5. Maximizing windows on the Pi should then work just fine. At least, it finally does for me.

jmlineb, 2022-07-22:
@yvessa, good hack!

romton843, 2022-07-28:
@jmlineb, When I try to uncomment dtoverlay=vc4-kms-v3d, I actually get trouble connecting through SSH indeed and even when I can, my VNC client shows the message “Cannot currently show the desktop”.
That’s why I said this solution doesn’t work for me. Thank you for responding though.

popcornmix, 2022-07-29:
Rather than trying a dummy hdmi plug, add `video=HDMI-A-1:1920x1080@60D` to the end of `/boot/cmdline.txt` (on same line). That should have the effect of faking a connected display.

romton843, 2022-08-10:
> Rather than trying a dummy hdmi plug, add `video=HDMI-A-1:1920x1080@60D` to the end of `/boot/cmdline.txt` (on same line). That should have the effect of faking a connected display.

I tried that but it actually made the headless stuck to that very small window. I'm guessing that a dummy hdmi plug will end up doing the same thing so I'm just going to reinstall it all. Thank you though.

Benik3, 2022-09-17:
For me forcing HDMI hotplug + resolution in /boot/config.txt AND adding the command to /boot/cmdline.txt fixed the issue.  

I can't disable kms because of camera. RPi must have "connected" display to get the VNC work properly. You must see the display under Preferences->Screen configuration

popcornmix, 2022-09-22:
> For me forcing HDMI hotplug + resolution in /boot/config.txt AND adding the command to /boot/cmdline.txt fixed the issue.

I'm surprised the config.txt settings made a difference when using kms driver.

Can you report exactly which settings you added to config.txt and can you try removing them one at a time to ensure which ones are really required.

Chippi, 2022-09-24:
> @romton843, I'm so sorry that the fix you mentioned is not working for you. But it gave me the idea to search the Web for that particular solution, and I found it, and it is working for me! Here are the details, inspired both by your comment and by https://forums.raspberrypi.com/viewtopic.php?t=282380.
> 
> 1. Comment out all dtoverlay=vc4-kms-v3d lines in config.txt
> 2. Comment out max_framebuffers too (but this might not be necessary)
> 3. Restart the Pi headless
> 4. Connect from another machine with VNC client. Best not to maximize the VNC client window but let the headless resolution that you previously set determine the window size. Otherwise you get some mouse anomalies trying to switch from window to window.
> 5. Maximizing windows on the Pi should then work just fine. At least, it finally does for me.
> 
> Some people have said that SSH access gets messed up with dtoverlay=vc4-kms-v3d commented out. When I get back home I'll test that out and see what is the case for me. Some other people have said that what you need instead is the older dtoverlay=vc4-fkms-v3d entry but that it will be deprecated soon. Your mileage may vary.

Warning don't do this, it might "brick" your PI. 

After trying this out, I had to:
- Remove my SD card
- Insert into PC
- Manually uncomment those lines
- Put SD card back into PI

jmlineb, 2022-09-26:
I'm so sorry, but what precisely bricked your Pi? Like I said, that approach worked for me.

But having said that, I'm currently a bit more nuanced in my config.txt. Here is a snippet of what I am now running with.

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Enable DRM VC4 V3D driver
[EDID=*]
# CONDITIONALLY ENABLES DTOVERLAY IF DISPLAY IS PLUGGED IN
dtoverlay=vc4-kms-v3d
# Set to number of monitors
max_framebuffers=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
gpu_mem=256

Benik3, 2022-09-26:
> I'm surprised the config.txt settings made a difference when using kms driver.
>
> Can you report exactly which settings you added to config.txt and can you try removing them one at a time to ensure [...]

I don't know why, but just editing config.txt worked for some time and after one reboot it stoped working. So I edited also the cmdline.txt and it started to work again...

@Chippi I also tried to disable kms and didn't "brick" the RPi, it must be something else...

Juraj-Masiar, 2022-12-02:
This problem appears on my RPi 4GB somehow randomly after each reboot.
But if logout and login back (which is faster than reboot), it may also fix itself (like 40% chance).

Isn't there a way to configure the "fallback resolution"? For me it's like 300px wide (not 1023 as mentioned above) so even logging out is pretty complicated.

AumPauskar, 2022-12-12:
You can change the headless resolution from the start menu itself to some other arbitrary number (in my case I tried using 1600x1200) then tried to reboot and then again set the display resolution to 1920x1080, while not touching raspi-config.

Juraj-Masiar, 2022-12-19:
> You can change the headless resolution from the start menu itself

Where in the menu?
Setting it in the Configuration window won't work on boot, I need to logout and login back every time in order to get a proper resolution.
[image]

This is what I get after reboot:
[image]

The usable screen is only the small rectangle in the upper left corner :(
[image]

japafrite, 2022-12-23:
I use Linux raspberrypi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

On 2 hardware versions (Pi 3B and Pi 3B+) :
- On Model : Raspberry Pi 3 Model B Rev 1.2 => NO display problem
- On Model : Raspberry Pi 3 Model B Plus Rev 1.3 => Display prolem

ahmedamoharram, 2023-02-03:
> Rather than trying a dummy hdmi plug, add `video=HDMI-A-1:1920x1080@60D` to the end of `/boot/cmdline.txt` (on same line). That should have the effect of faking a connected display.

This actually worked in my case, but looks like it only supports standard resolutions, those worked with me:
1280x720
1600x900
1920x1080

ramit99, 2023-04-07:
Very interested in this thread.

I've tinkered with this, and have screwed up the screen resolution, had to put it on a monitor and m/k to fix things
 
So I've had this exact same issue for a year now.. just the luck of the draw on a reboot or cold start if the desktop will act like the resolution it's set at, or something smaller. I've had it on a Pi3+ 32b OS, and Pi4b+ 64b OS . All via VNC.
I've set the headless res to 1920x1080, even checking it in the cl config panel for VNC display option.
I tried this,,   video=HDMI-A-1:1920x1080@60D to the end of /boot/cmdline.txt  , NG for me.
About to get the HDMI dummy plug. this is really annoying,

ramit99, 2023-04-07:
I was just doing some googling on this, and found some input, and on my 5th reboot, and all is hanging in there..

found in thread here:
https://forums.raspberrypi.com/viewtopic.php?t=332327

/boot/config.txt
Clipped from working config.txt...
First 3 lines were in my stock config.txt , lines following are what were added.

`# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

hdmi_group=2
hdmi_mode=82

# Ignore EDID display data
hdmi_ignore_edid=0xa5000080
hdmi_force_hotplug=1`

ramit99, 2023-04-07:
I started with the same lines as the Pi4B , but didn't work.
Lots of adding and changing code for the resolution and rem'ing out , and so 5 reboots on a Pi3b+ 32b os.. and it's staying !

As with all, have the headless set to 1920x1080.

5 reboots and with this in the config.txt, all is good...

```
# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=5

hdmi_group=2
hdmi_mode=82

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2
```
Going to sleep.. 

Figures, I ordered the dummy hdmi's, and finally find combo's that work after hacking and wacking keys.

iRamy, 2023-04-09:
Hi, for those that could find this useful. What worked for me to fix the maximizing of the windows is this :

- Go to the config.txt file : sudo nano /boot/config.txt
- Uncomment these two lines and set the desired resolution :

framebuffer_width=1920
framebuffer_height=1080

- Save and reboot

I hope this helps

hystrix1, 2023-04-15:
Setting framebuffer_width=1920 and framebuffer_height=1080 in config.txt worked for me. Many thanks.

ZigaJavornik, 2023-04-25:
I had the same issue, here is the solution that worked for me:

in the /boot/config.txt edit this, make sure it is uncommented - after a reboot, you will see maximized windows, for me at least it was the case :
hdmi_group=2
hdmi_mode=82

MKoen, 2023-04-27:
@ZigaJavornik, thank you! Setting `hdmi_group=2` and `hdmi_mode=82` fixed it for me.