raspberrypi/Raspberry-Pi-OS-64bit #226: USB3 Camera need to physically reconnect after reboot to work

mv4manishverma, 2022-05-09:
Hi,
This is the similar problem mentioned on this link.
[https://forums.raspberrypi.com/viewtopic.php?t=320242](https://forums.raspberrypi.com/viewtopic.php?t=320242)
I have a 2.5W 5V USB3.0 Camera. I tried to use it on both raspberry pi 4B and cm4 (on rasp pi io board thorough PCIE to USB3 adpator). In both cases, after rebooting the pi, the camera does not work and I have to physically disconnect and reconnect to make it work.
After rebooting, Dmesg does not show any issues and lsusb also shows the camera correctly. I am not sure what is the issue.
As mentioned in the above link, I have to use another USB3 hub (powered or not does not matter, both works) between pi and the camera to make it work. Also, i am not convinced that it is because of any power issue (camera drawing more current).

lurch, 2022-05-09:
Perhaps there's some "internal state" in the camera that needs a full power-disconnect to reset, and doesn't get properly reset when the kernel re-enumerates the USB devices during bootup? Perhaps this issue belongs in https://github.com/raspberrypi/linux ?

mv4manishverma, 2022-05-09:
> Perhaps there's some "internal state" in the camera that needs a full power-disconnect to reset, and doesn't get properly reset when the kernel re-enumerates the USB devices during bootup? Perhaps this issue belongs in https://github.com/raspberrypi/linux ?

It work correctly with windows even after reboot. So I suppose the problem is with Pi?

mv4manishverma, 2022-05-09:
> Perhaps there's some "internal state" in the camera that needs a full power-disconnect to reset, and doesn't get properly reset when the kernel re-enumerates the USB devices during bootup? Perhaps this issue belongs in https://github.com/raspberrypi/linux ?

I just found that this issue also exists  with Linux PC running Ubuntu 20.4.4. So, as you said the issue might be with the kernel.

JamesH65, 2022-05-09:
Almost certainly a problem with the Linux driver, as now shown by it happening on Ubuntu. Might be worth trying the latest Ubuntu, it might have been fixed in a more recent kernel, which will make it easier to find the issue and backport.

6by9, 2022-05-09:
> In both cases, after rebooting the pi, the camera does not work

> After rebooting, Dmesg does not show any issues and lsusb also shows the camera correctly.

Contradictory.

Does the device show up after reboot? If so, then compare the output from ```sudo lsusb -vvv``` before and after unplugging and replugging.

6by9, 2022-05-09:
Also module driver uvcvideo (and a few dependencies should be loaded if the device is detected correctly (assuming it's a standard USB Video Class device).

mv4manishverma, 2022-05-10:
> 

> Contradictory.
>
> Does the device show up after reboot? If so, then compare the output from `sudo lsusb -vvv` before and after unplugging and replugging.

Yes the device shows up on reboot correctly. I checked the output of `lsusb -vvv` before and after reboot. The details of the device is same. I checked it with the `diff` command.

mv4manishverma, 2022-05-10:
> Also module driver uvcvideo (and a few dependencies should be loaded if the device is detected correctly (assuming it's a standard USB Video Class device).

The camera is not a webcam. It is similar to a scientific camera.

lurch, 2022-05-10:
> Yes the device shows up on reboot correctly. I checked the output of lsusb -vvv before and after reboot.

Perhaps you could give more details about "the camera does not work" ? With the little information you've provided so far, it's pretty hard to guess at what the problem might be.

mv4manishverma, 2022-05-10:
> > Yes the device shows up on reboot correctly. I checked the output of lsusb -vvv before and after reboot.
> 
> Perhaps you could give more details about "the camera does not work" ? With the little information you've provided so far, it's pretty hard to guess at what the problem might be.

Hi, I found a similar issue on the internet.
[https://github.com/basler/pypylon/issues/347](https://github.com/basler/pypylon/issues/347)

I am attaching the output of all the commands you mentioned, you can go through it and let me know if you find something.
[dmesg_AfterReboot.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663434/dmesg_AfterReboot.txt)
[dmesg_freshlyConnected.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663435/dmesg_freshlyConnected.txt)
[lsmod_AfterConnectingCamera.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663436/lsmod_AfterConnectingCamera.txt)
[lsmod_AfterReboot.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663437/lsmod_AfterReboot.txt)
[lsusb_AfterReboot.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663438/lsusb_AfterReboot.txt)
[lsusb_freshlyConnected.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/8663439/lsusb_freshlyConnected.txt)

mv4manishverma, 2022-05-12:
Hi all,
Thanks for your support on this issue. The problem is solved with a firmware update for the camera from the manufacturer.
I am closing this issue now.

closed 2022-05-12 by mv4manishverma