raspberrypi/Raspberry-Pi-OS-64bit #227: ssh default passwd

rmGFW, 2022-05-20:


rmGFW, 2022-05-20:
command:ssh pi@raspberrypi.local
passwd:raspberry

popcornmix, 2022-05-20:
See https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

lurch, 2022-05-20:
And also https://www.raspberrypi.com/documentation/computers/getting-started.html#configuration-on-first-boot especially the part where it says:

> If you are installing Raspberry Pi OS Lite you will still need to create a new user account. You will therefore be prompted to create an account by text prompts at the command line when you first boot a Lite image. If you are booting Raspberry Pi OS [headless](https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi) you MUST configure the operating system using Raspberry Pi Imager using the [Advanced Menu](https://www.raspberrypi.com/documentation/computers/getting-started.html#advanced-options).

mikenor, 2022-05-22:
> especially the part where it says:

That is incorrect. Using the Imager tool is NOT a MUST. The link shared earlier by @popcornmix explains:

> There are also mechanisms to preconfigure an image without using Imager. To set up a user on first boot and bypass the wizard completely, create a file called userconf or userconf.txt in the boot partition of the SD card....

I can confirm that the `/boot/userconf` file does work - no Imager required.

lurch, 2022-05-23:
Thanks - I've added a note to https://github.com/raspberrypi/documentation/issues/2537