raspberrypi/Raspberry-Pi-OS-64bit #228: Kernel panic caused by bcm2835_spi_interupt

wbb12, 2022-05-25:
When I use the Raspberry Pi 4B with raspberry pi OS 64 bits, the bcm2835_spi_interupt always causes the kernel to panic.  Two SPIs are used on the Pi, spi0 for mcp2515 and spi6 for ADS1263. In the beginning, two SPI devices work correctly, but the kernel panic always occurs after several minutes or several minutes. Actually, I installed the ubuntu server 22.04 64bits on another Pi and it shows the same problem.

These are the messages show on the monitor. 
[image]
[image]

```
$ uname -a
Linux rpi-test1 5.15.0-1008-raspi #8-Ubuntu SMP PREEMPT Thu May 19 11:58:51 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
$ uname -a
Linux rpi-test1 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
Both the ubuntu and raspberry pi OS are 5.15 kernel.

The SPIs devices are working well on the Pi with 32bits OS.

lurch, 2022-05-25:
This is clearly a kernel bug, rather than an OS-level bug, so should probably be reported at https://github.com/raspberrypi/linux instead?

wbb12, 2022-05-25:
Thank you! I will report it to  https://github.com/raspberrypi/linux.
王彬彬
SUNIST实验室
工程物理系
清华大学
***@***.***
Mobile:+8618811362084