raspberrypi/Raspberry-Pi-OS-64bit #229: kodi 32 bit

molli1312, 2022-05-29:
I took the chance to try the newly released 64bit Raspberry Pi OS bullseye. While initial setup worked fine and also installation of kodi works fine. there is no 64bit widevine CDM available to use plugins like netflix via inputstream.adaptive. Just like install a 32bit kodi. But I cant.

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
kodi:armhf : Depends: kodi-bin:armhf (>= 2:19.1+dfsg2-2+deb11u1) but it is not going to be installed
Depends: kodi-bin:armhf (< 2:19.1+dfsg2-2+deb11u1.1~) but it is not going to be installed
Recommends: kodi-repository-kodi:armhf or
kodi-repository:armhf
Recommends: kodi-visualization-spectrum:armhf but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

popcornmix, 2022-05-31:
I've checked and 32-bit kodi is not currently supported on RPiOS 64-bit.
It seems the packaging needs work to support multiarch.

molli1312, 2022-05-31:
Can i hope that anybody fix this problem in the future

lurch, 2022-06-05:
That might be a question for the Kodi developers rather than the Raspberry Pi OS developers? :shrug:

molli1312, 2022-06-05:
The kodi devolpers said "You'll need to contact your distro about those options. It's not handled by us", because kodi 32 bit is available...

lurch, 2022-06-05:
Ahhh, in that case you might need to ask the Debian maintainers, because Raspberry Pi OS 64-bit takes packages directly from upstream Debian. https://packages.debian.org/bullseye/kodi

XECDesign, 2022-06-05:
> Ahhh, in that case you might need to ask the Debian maintainers, because Raspberry Pi OS 64-bit takes packages directly from upstream Debian. https://packages.debian.org/bullseye/kodi

We provide our own packages. They just don’t support multiarch

gamaralf, 2022-10-21:
> We provide our own packages. They just don’t support multiarch

@XECDesign , what is the URL for me to browser the packages you provide? Thanks in advance.

lurch, 2022-10-22:
https://archive.raspberrypi.org/debian/dists/bullseye/

Schlumpf7, 2023-02-17:
I'm interested in this, too
I also tried to setup a schroot armhf environment for Kodi:armhf. But I can't get mouse an keyboard to work in there.
A package for a multiarch environment would be much easier.

popcornmix, 2023-02-17:
Do you only need multi-arch for 32-bit widevine?
If so, there should be a solution for 64-bit kodi + widevine soon.

Schlumpf7, 2023-02-17:
You got me there. Sadly arm64 lacks on support for widevine. My workaround was to install the armhf version. 
I guess kodi:arm64 with widevine will perform much better than my schroot enviorment.

popcornmix, 2023-02-17:
Keep an eye on https://github.com/xbmc/inputstream.adaptive/pull/1130
If a solution is merged there, we can rebuild the inputstream.adaptive addon and 64-bit kodi will have widevine support.

If there isn't a solution there soon, we may be able to solve it with LD_PRELOAD.

lockie, 2023-02-25:
The MR was merged. Any news on this?
I'm kinda tired fighting dumb Debian package manager, just to watch a little Netflix
```
root@balthasar:~# apt-get install --no-install-recommends -f kodi:armhf=2:19.1+dfsg2-2+deb11u1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kodi-bin:armhf : Depends: libsmbclient:armhf (>= 2:4.0.3+dfsg1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

root@balthasar:~# apt install libsmbclient:armhf
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libsmbclient:armhf is already the newest version (2:4.13.13+dfsg-1~deb11u5).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```

pelwell, 2023-02-25:
It's less than 12 hours since the merge, and it's the weekend, and you're expecting it to have been incorporated and packaged and released already?

lockie, 2023-02-25:
@pelwell sorry, I've had rough week (well, year really) and just really pissed by necessity to spend my weekend fixing Linux packaging problems instead of finally watching myself some Wednesday

lockie, 2023-02-25:
Ah, NVM, I was able to install and run 32-bit (armhf) Kodi version on my RPi; for some weird reason `bullseye-backports` repo has lower priority than the `bullseye`. For anyone interested, the incantation was
```
apt install kodi-bin:armhf=2:19.4+dfsg2-2~bpo11+1 kodi:armhf=2:19.4+dfsg2-2~bpo11+1 kodi-data:=2:19.4+dfsg2-2~bpo11+1
```
So sorry for the fuss. Have a great day everyone.