raspberrypi/Raspberry-Pi-OS-64bit #230: No bluetooth hci0 interface getting detected for Raspberry Pi Compute Module 4.

indrajeetshirole2407, 2022-06-03:
We are tiring work on Bluetooth on Raspberry Pi Compute Module 4. We are running Poky (Yocto Project Reference Distro) 3.1.13 on Raspberry Pi Compute Module 4. We are using dunfell branch for meta-raspberrypi.

We are not able to see the bluetooth interface hci0 if I type _hciconfig_ command.
```
root@wacout:/# hciconfig
root@wacout:/#
```

I don't see any devices if I type _hcitool dev_ command

```
root@wacout:/# hcitool dev
Devices:
root@wacout:/#
```

If I try to run _systemctl start hciuart.service_ command, it fails with Initialization timeout error.

```
Jun 03 10:03:10 wacout systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jun 03 10:03:40 wacout btuart[65882]: Initialization timed out.
Jun 03 10:03:40 wacout btuart[65882]: bcm43xx_init
Jun 03 10:03:40 wacout systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mhciuart.service: Control process exited, code=exited, status=1/FAILURE[[0
m
Jun 03 10:03:40 wacout systemd[1]: [[0;1;38;5;185m[[0;1;39m[[0;1;38;5;185mhciuart.service: Failed with result 'exit-code'.[[0m
Jun 03 10:03:40 wacout systemd[1]: [[0;1;31m[[0;1;39m[[0;1;31mFailed to start Configure Bluetooth Modems connected by UART.[[0m

```
Any idea how to fix these error?

I will highly appreciate your help

pelwell, 2022-06-03:
Does ` bluetoothctl power on` help?

indrajeetshirole2407, 2022-06-04:
> Does ` bluetoothctl power on` help?

Thanks for the reply, I ran the command but it says no  default controller available
```
root@wacout:~# bluetoothctl power on
No default controller available
```

dmitriychernenko, 2023-06-18:
Experienced exactly the same issue.

Did you try to update your rpi firmware?
run `sudo rpi-update` and confirm when it asks you "Would you like to proceed?"

Reboot and voila:
`xxxxxxx@raspberrypi:~ $ hcitool dev
Devices:
        hci0    XX:XX:XX:XX:XX:XX`